rm-hull/luma.oled #278: Missing row on SSD1306

gilphilbert, 2020-03-09:
#### Type of Raspberry Pi
Raspberry Pi 2b 1.1

#### Linux Kernel version
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux

####
MakerFocus i2c OLED display
https://www.amazon.com/gp/product/B0761LV1SD

#### Expected behaviour
Full display across all rows

#### Actual behaviour
The top row of the display isn't shown, so in a basic "draw a box" to the size of the bounds, the top line of the box is missing. I managed to make it reappear by changing the bounds of the image:

```
from luma.core.interface.serial import i2c 
from luma.core.render import canvas
from luma.oled.device import ssd1306

serial = i2c(port=1, address=0x3C)
device = ssd1306(serial,128,64,0)

#set the bounds of the box to remove the top row
box = (0, 1, 127, 63)

with canvas(device) as draw:
    draw.rectangle(box, outline="white", fill="black")
    draw.text((0, 0), "Hello World", fill="white")
```

The display is one I bought from Amazon (probably a mistake, in hindsight...) and the datasheet is here: 
https://drive.google.com/file/d/14A_luFwqvZaKu-PKTfJdWmUZ4T4Wk65X/view
The chip is stated as an SSD1306G

Any help appreciated!

rm-hull, 2020-03-09:
For other devices, we have implemented a parameter called **h_offset** to countenance exactly what you are describing. Usually, it is when the memory is not properly mapped to the first display line for whatever reason.

We have never had to do it for SSD1306 before though. Could you get hold of the luma.examples repository and run the demo.py script and take a picture and attach it to this issue?

gilphilbert, 2020-03-12:
OK, so it's user error. I didn't notice I had the screen height set to 64 instead of 32. Explains why it looked so strange...

closed 2020-03-12 by gilphilbert