rm-hull/luma.oled #279: SSD1322 shows only white noise

majuss, 2020-03-23:
Hi guys, I'am running an SSD1322 (https://de.aliexpress.com/item/32949282762.html?spm=a2g0s.9042311.0.0.26e74c4dSrLkVM) on a Pi 3 B+. Everything worked for a few tries, then it got fishy. The display didn't respond to commands, colors (monochrome) were inverted and so on. Now the perftest (from luma.examples) shows an insanely high fps number and with the hello world example I only get white noise... What are ways to debug this? Could I have killed the display while touching wires and maybe disconnecting them while running? Thank you for the help!

rm-hull, 2020-03-24:
It’s possible you have fried the display and/or the gpio pins on the Rpi, but I think it is unlikely since you look to be using female connector wires so it should be impossible to touch the wires. If you have wired up ground and VCC the wrong way, then this may have shorted the displays breakout board. The only thing I can think is to double check your wiring, try another RPi if you have one, or buy a new display.

Perftest will show a high FPS by default because this driver does an optimization if the image is unchanging. This is unrelated to your issue.

majuss, 2020-09-08:
The Display itself was still good. Somehow the electronic components on the board acted up... Switch the board and the display works well now.

closed 2020-09-08 by majuss

rm-hull, 2020-09-08:
Glad you managed to get it sorted out :+1: