rm-hull/luma.oled #280: Text being bleugh... Antialiasing?

nigeljohnson73, 2020-03-29:
I'm trying to upgrade a project from a 0.96" (128x64) oled using the Adafruit library to a 1.5" (128x128) using Luma. I have the same number of pixels across, so I assume they are just bigger on the bigger display.

When I load a font at 12pt on the small display, the edges are crisp and are a single pixel thick, and all looks good. On the bigger display, they are soft and mushy and seem to be trying to scale down a bigger font?

I am using the same font load line and the same draw.text() call, the only difference I can see is the way the PIL image is created, small display  - manually; Luma - using a canvas call... as well as the fact I'm using a display now that has some greyscales.

How do I switch off the antialiasing call when rendering text? I still want the greyscale if I'm drawing a bit later? I found something on putData and getting a mask, but I got a bit confused - I'm a python noob.

The first image is showing what I have now with 12 and 10pt fonts rendering really nicely. The other 2 images are of a 10 and 12pt font using the bigger display. You can see the corners are darker so looks antialiased.

[image]
[image]
[image]

rm-hull, 2020-03-30:
Which font are you using? You could try using a bitmap font rather than a TTF font.

Else you could try using one of the legacy fonts - see https://luma-core.readthedocs.io/en/latest/legacy.html, and here for example use: https://github.com/rm-hull/luma.examples/blob/afae00a7aa760bff80b9ff87a74feb61f985cf6f/examples/weather.py#L39

nigeljohnson73, 2020-03-30:
I am using andale mono. Is there an example using the `with canvas` I've got quite a bit to rewrite if I'm using show_message with no control

rm-hull, 2020-03-30:
Sorry wrong method ... There is a text method in the legacy namespace ... https://luma-core.readthedocs.io/en/latest/legacy.html#luma.core.legacy.text

That is what I meant rather than show message.

nigeljohnson73, 2020-03-30:
Wow, they are horrid :D LCD_FONT seems ok though, and the date just fits across the display! Thanks. :D 

There must be a way to revert back to the functionality of a worse screen as readability is a lot worse if you want to see it bigger. Can I add that as a feature request?

rm-hull, 2020-03-30:
you can make any of the legacy bitmap fonts proportional rather than fixed width by wrapping it with `proportional(LCD_FONT)` ... That should make it more palatable.

You can use the pillow drawing primitives with certain bitmap style TTF fonts - see here for example: https://github.com/rm-hull/TR4-monitor/blob/master/tr4_monitor/fonts/__init__.py

> Can I add that as a feature request?

Remember at this point, it is a feature of pillow that you are actually using to render onto an image. All this library does is take the resultant images and convert them the correct sequences of commands and data for the display.

nigeljohnson73, 2020-03-31:
I love the mono look - andale mono, but I will give it a whirl. :) thanks

But the pillow feature (which I assume is the same regardless of which library I'm using) renders the same TTF file in the same number of pixels (128) across the screen depending on whether the display can anti-alias or not... I would like the same behaviour for drawing text on the 1.5" screen using luma as I get on the 0.96" using the Adafruit library.

The only difference I expected with the 2 displays is 64 extra pixels high, and a bigger hole.

rm-hull, 2020-03-31:
Well, the image you are drawing onto has different characteristics .. the one for the smaller screen is initialized to a 1-bit color depth while the other is RGB. To be fair, I don’t even know which drivers you are using, so am assuming the 1.5” screen is one that supports greyscale.

If you initialize the 1.5” screen with mode=“1”, and draw text on it, it won’t do the anti-aliasing, but I seem to recall you said you didn’t want that in your original comment.

Having said that, I’ve just googled and found this: https://mail.python.org/pipermail/image-sig/2005-August/003497.html ... worth a try?

nigeljohnson73, 2020-04-01:
I did try the mode=1 but it didn't seem to do anything. fontmode however, that is pretty darned close. It's still thicker than one pixel, so there is still something else going on, but I'm gonna be happy with that.

Thanks for all your help.

nigeljohnson73, 2020-04-01:
Oh, sorry, yeah the 0.96-inch display is 1bit SSD1306 based and using the Adafruit library, the 1.5 inch is 16 greys I think, and using luma. both on python3 and whatever version of PIL that comes with on the PI.

closed 2020-04-01 by nigeljohnson73