rm-hull/luma.oled #281: regulate brightness (contrast) of ssd1306 display

sybednar, 2020-04-07:
Hi
This is my time posting a question on github so hopefully I am doing this correctly.

First, I would like to thank all of the contributors in generating the Luma display package. I am a newbie to working with RPIs and based on the documentation I was able to get the code working wonderfully for my application.

Background: I am using a ssd1306 OLED 128 x 64 display running on a RPI 3B+
linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15;58 GMT 2020 armv7l GNU/Linux

Objective: Vary brightness (contrast) of 1.54" ssd1306 OLED display. Display is embedded in prototype chassis and is working appropriately so I would prefer to not have to change display.  However I did not realize how bright these displays are, so now I would like to add code to vary brightness(contrast)

Problem:  Running the simple program below over a range of  contrast values ( 1-255) has no affect on display brightness.  I have tested code on various other ssd1306 oled displays including i2c and spi versions from 3 other vendors and likewise see no change in brightness when changing Contrast value.  However, changing  contrast value on sh1106 oled display  works fine (from dim to very bright).

Question:  What am I doing wrong? Alternatively, is there additional code required for varying the contrast of ssd1306 driven displays?
According to postings on forums where others have used ssd1306 displays (arduino code) and varied the brightness it is necessary to vary the  pre charge time.  Is this feasible using the luma oled code? If so, I would greatly appreciate your suggestions (for a relatively coding newbie) on how to implement.

Thanks

Simple test program:
from luma.core.interface .serial import i2c, spi
from luma.core.render import canvas
from luma.oled.device import ssd1306, ssd1325, ssd1331, sh1106
serial =spi(device=0, port=0)

Contrast =10  #range 0-255
device.contrast(Contrast)

try:
while True:
with canvas(device) as draw:
draw.rectangle(device.bounding_box, outline="white", fill="black")
draw.text((30,10), "Hello World", fill= "white")''

finally:
device.cleanup()

> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

rm-hull, 2020-06-03:
setting contrast as you have done should work, if you try and run the https://github.com/rm-hull/luma.examples/blob/master/examples/demo.py that should regulate the brightness output  - does that work for you?

rm-hull, 2020-06-03:
It would be useful if you can provide a link to the OLED breakout board you are using also

closed 2023-06-11 by thijstriemstra