rm-hull/luma.oled #283: draw instance doesn't always render when it goes out of scope

nathan-stewart2, 2020-04-21:
This is likely a usage issue but I can't find anything dealing with it:

I've written a simple program and the text I render does appear on the display when the draw object goes out of scope:
with canvas(device) as draw:
   draw.text(pos, "This works"....)

But I then created a callback function, which is being called correctly. I've verified that the oled device instance is the same - I can either use it globally or pass it in as callback data. So far so good. But then the attempt at drawing doesn't render when it goes out of scope. I've added logging to show that a new draw object is being created and going out of scope, but nothing appears on the display:### 

font = ImageFont.truetype('/usr/share/fonts/truetype/dejavu/DejaVuSans.ttf', 8)
with canvas(oled) as draw:
    oled.clear()
    draw.text((0, 0),"This text works",  font=font, fill="white")

def song_changed(event, oled):
     print("This callback is being called")
     with canvas(oled) as draw:           
          print("draw instance = %s" % draw)
          draw.text((0, 0), "But This text does not show", font=font, fill="white")

event_manager.event_attach(vlc.EventType.MediaListPlayerNextItemSet, song_changed, oled)

I'm sure this has to have something to do with the way it's being called but all my attempts to trace it appear as if it should be working. What am I doing wrong?

rm-hull, 2020-04-22:
Firstly, you don’t need to call ‘oled.clear()’ inside the ‘with canvas‘ block.

I can’t see why your code wouldn’t work. Presumably if you were to call ‘song_changed(event, oled)’ directly, it works?

If that is the case, then I would look at the docs for event manager’s event_attach method more carefully to check how parameters are passed through.

BTW what class/package is the event manager from?

nathan-stewart2, 2020-04-22:
I've done some additional tracing. It appears to work until it launches the RotaryEncoder class - which is GPIO only, not SPI. So I'm thinking the encoder class is breaking something in the hardware access. I'm also getting ready to replace the python gpio bit banging encoder with an I2C encoder board with interrupts - so I'm going to just punt on this one.

nathan-stewart2, 2020-04-22:
Closing this, one of my libraries was yanking pin 11 out of SPI mode.

closed 2020-04-22 by nathan-stewart2