rm-hull/luma.oled #284: AssertionError with SSD1309

Maschine2501, 2020-04-27:
Hi, 
I got an "AssertionError" with an SSD1309 Display.
The Situation:
I got a Python Code for the SSD1322 running [SSD1322-Code](https://github.com/Maschine2501/NR1-UI), which is working fine.

Now I wanted to port it to an SSD1309 Display [SSD1309-Code](https://github.com/Maschine2501/RC2-UI), and keep getting an Assertion Error.

Here's the log:
Apr 27 07:28:48 raspberrypi systemd[1]: Started RC2-UI.
Apr 27 07:28:53 raspberrypi rc2ui-nightly[825]: string of image.mode in __init__.py  1
Apr 27 07:28:53 raspberrypi rc2ui-nightly[825]: string of self.mode in __init__.py  1
Apr 27 07:28:53 raspberrypi rc2ui-nightly[825]: string of image.mode in __init__.py  1
Apr 27 07:28:53 raspberrypi rc2ui-nightly[825]: string of self.mode in __init__.py  1
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]: Cannot open file volumio_logo.ppm
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]: string of image.mode in __init__.py  RGB
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]: string of self.mode in __init__.py  1
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]: Traceback (most recent call last):
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]:   File "/home/pi/RC2-UI/rc2ui.py", line 873, in <module>
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]:     show_logo("volumio_logo.ppm", oled)
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]:   File "/home/pi/RC2-UI/modules/display.py", line 15, in show_logo
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]:     device.display(logoImage)
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]:   File "/home/pi/.local/lib/python3.7/site-packages/luma/oled/device/__init__.py", line 202, in display
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]:     assert(image.mode == self.mode)
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]: AssertionError
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]: string of image.mode in __init__.py  1
Apr 27 07:28:54 raspberrypi rc2ui-nightly[825]: string of self.mode in __init__.py  1
Apr 27 07:28:55 raspberrypi systemd[1]: rc2ui.service: Main process exited, code=exited, status=1/FAILURE
Apr 27 07:28:55 raspberrypi systemd[1]: rc2ui.service: Failed with result 'exit-code'.

My Code has a [main-script](https://github.com/Maschine2501/RC2-UI/blob/master/rc2ui.py) which tells a [submodule](https://github.com/Maschine2501/RC2-UI/blob/master/modules/display.py) to display a logo at startup. 
I've tried it with .ppm .bpm and .jpg files, error persists.
To figure out what happend i added 2 prionts in __init__.py: "string of image.mode" and "string of self.mode" right before line 196 in [__init__.py](https://github.com/rm-hull/luma.oled/blob/master/luma/oled/device/__init__.py), to see what is given there.

In my display [submodule](https://github.com/Maschine2501/RC2-UI/blob/master/modules/display.py) i start the image processing to luma with:

def show_logo(filename, device):
    logoImage = Image.new('RGB', (device.width, device.height))

In the [API Documentation](https://luma-oled.readthedocs.io/en/latest/api-documentation.html) is stated that "mode" (1, RGB or RGBA) is supported since Luma 3.1 -> I'm using 3.4.

In the [__init__.py](https://github.com/rm-hull/luma.oled/blob/master/luma/oled/device/__init__.py) ssd1309 is redirected to ssd1306. ssd1306 has a different opener than ssd1322, which is stated to be a "greyscale" device.

My Hardware:
-PiZeroWH (Raspbian Buster Lite)
-ssd1309 SPI Display (Resistors solderd correctly to use spi)

Maybe anyone has an Idea what I am doing wrong?

Cheers!

mindforger, 2020-04-27:
from the looks of it the 1309 has no "mode" to be configured or mayybe it has not been added yet and seems to fallback to monochrome in that case

Maschine2501, 2020-04-27:
> 
> 
> from the looks of it the 1309 has no "mode" to be configured or mayybe it has not been added yet and seems to fallback to monochrome in that case

That seems to be true, but why is it stated in the API to support "modes"?
-> Do you have an idea how i must change my code [display.py](https://github.com/Maschine2501/RC2-UI/blob/master/modules/display.py) so that it don't use a mode?

I've installed luma.examples to test the display... Even the examples don't display anything.
But the Error has changed since then:

on3.7/site-packages/luma/oled/device/__init__.py", line 198, in display
Apr 27 16:00:49 raspberrypi rc2ui-nightly[698]:     assert(image.mode == self.mo                                                                                                                                                             de)
Apr 27 16:00:49 raspberrypi rc2ui-nightly[698]: AttributeError: 'tuple' object h                                                                                                                                                             as no attribute 'mode'

The testfile:
 python3 3d_box.py --display ssd1309 --interface spi
Version: luma.oled 3.4.0 (luma.core 1.14.0)
Display: ssd1309
Interface: spi
Dimensions: 128 x 64
------------------------------------------------------------

mindforger, 2020-04-27:
you can use a mode (although i did not find the point where it is defined by default or could not trace the inherited base class) but you have to go monochrome if it does not support physically RGB i guess

you can always check the mode of the render device object before your issue a rendering of a picture in a certain mode

rm-hull, 2020-04-27:
If you use the display method directly, it is the callers responsibility to set the correct mode (bit depth) that the device supports. The canvas will always construct one which is correct to match the device, but display doesnt.

See here how to convert: https://pillow.readthedocs.io/en/stable/reference/Image.html#PIL.Image.Image.convert

You can get the target mode from device.mode

Also see: 
- https://github.com/rm-hull/luma.examples/blob/master/examples/invaders.py#L202
- https://github.com/rm-hull/luma.examples/blob/master/examples/pi_logo.py#L30

Maschine2501, 2020-04-29:
Hey Guys! Thank you for your Reply's!
That my display doesn't show anything -> was my fault, bad cables...

With your advice ".convert"... i was able to solve my problem. 

Thank you!!!!

closed 2020-04-29 by Maschine2501