rm-hull/luma.oled #286: Python 3 + sh1106 + SPI interface = the RAM is used more and more until the system kills the process.

darton, 2020-05-04:
When I use python 3 (3.5.3) with the sh1106 device with the SPI interface, the RAM is used more and more until the system kills the process.

When I use python 2.7 the problem does not occur, the problem does not occur when I use python 3 with the sh1106  on i2c interface

----------
RPi 3 Hardware        : BCM2835
Revision        : a020d3

Linux rpims_z1.wisniowski.pl 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

sudo -H pip3 install --upgrade luma.oled
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: luma.oled in /usr/local/lib/python3.5/dist-packages (3.4.0)
Requirement already satisfied, skipping upgrade: luma.core>=1.12.0 in /usr/local/lib/python3.5/dist-packages (from luma.oled) (1.12.0)
Requirement already satisfied, skipping upgrade: pyftdi; python_version >= "3.5" in /usr/local/lib/python3.5/dist-packages (from luma.core>=1.12.0->luma.oled) (0.29.4)
Requirement already satisfied, skipping upgrade: pillow>=4.0.0 in /usr/lib/python3/dist-packages (from luma.core>=1.12.0->luma.oled) (4.0.0)
Requirement already satisfied, skipping upgrade: RPI.GPIO; platform_system == "Linux" in /usr/lib/python3/dist-packages (from luma.core>=1.12.0->luma.oled) (0.6.5)
Requirement already satisfied, skipping upgrade: smbus2 in /usr/local/lib/python3.5/dist-packages (from luma.core>=1.12.0->luma.oled) (0.2.3)
Requirement already satisfied, skipping upgrade: spidev; platform_system == "Linux" in /usr/lib/python3/dist-packages (from luma.core>=1.12.0->luma.oled) (3.3)
Requirement already satisfied, skipping upgrade: pyusb>=1.0.0 in /usr/local/lib/python3.5/dist-packages (from pyftdi; python_version >= "3.5"->luma.core>=1.12.0->luma.oled) (1.0.2)
Requirement already satisfied, skipping upgrade: pyserial>=3.0 in /usr/lib/python3/dist-packages (from pyftdi; python_version >= "3.5"->luma.core>=1.12.0->luma.oled) (3.2.1)

rm-hull, 2020-05-05:
Can you share the program that causes this memory leak?
What logging do you see for you to draw this conclusion?

From what I deduce, it seems like SPI on python3 may be the culprit .. have you looked at any issues on https://github.com/doceme/py-spidev (the SPI library we use) that may be responsible?

darton, 2020-05-06:
This program running in python 3 causes memory leakage.

https://github.com/darton/RPiMS/blob/master/test/luma_oled-test.py

To find out that the program causes memory leaks, I just watched its memory resources in the Top program. Memory leaks can be seen after a few minutes.
After several hours of operation, the program has all available memory.

pi@rasrpberry:~ $ dmesg

[pią maj  1 12:27:13 2020] Out of memory: Kill process 394 (python3) score 848 or sacrifice child
[pią maj  1 12:27:13 2020] Killed process 394 (python3) total-vm:967944kB, anon-rss:804640kB, file-rss:0kB, shmem-rss:0kB
[pią maj  1 12:27:13 2020] oom_reaper: reaped process 394 (python3), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB

darton, 2020-05-07:
Everything seems to indicate that SPI on Python 3 with sh1106 device is responsible for memory leaks.
SPI + Python3 + st7735 working OK.

thijstriemstra, 2020-05-08:
> When I use python 3 (3.5.3) 

That's quite old. Can you try latest 3.5.x or even better, 3.7 or 3.8?

darton, 2020-05-11:
You were right.
Upgrading Python to version 3.7.3 has solved the problem.

thijstriemstra, 2020-05-11:
Nice!!

closed 2020-05-11 by thijstriemstra