rm-hull/luma.oled #290: Adding support for 128x96 (1.27") Adafruit ssd1351

gh123man, 2020-07-06:
#### Raspberry Pi Zero W
#### Linux Kernel version
Linux raspberrypi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux
#### Expected behaviour
Display should function with the ssd1351 driver
#### Actual behaviour
Nothing. 

### Details

I am trying to get luma.oled working for this devuce: https://www.adafruit.com/product/1673 
It seems like there is good support for the 128x128 variant however even after adding the 128x96 resolution here: https://github.com/rm-hull/luma.oled/blob/master/luma/oled/device/__init__.py#L367 the screen does not respond. 

I have tested this screen with the Adafruit reference library https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display so I am confident everything is wired correctly and am able to talk to the device. The device specific parameters can be found here: https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display/blob/master/examples/rgb_display_pillow_demo.py#L36 
I also noticed some minor differences in the init process. I attempted to replicate the exact init sequence in luma.oled but with no luck. 

This all is rather new to me. I am happy to author support for this screen, but I am looking for some pointers on where I may be going wrong and how to further diagnose this.

gh123man, 2020-07-06:
I spoke too soon. 
issue was the GPIO pin placement and the defaults this library uses (learned from this thread https://github.com/rm-hull/luma.oled/issues/205)

Ill post a PR later to add the new resolution.

closed 2020-07-06 by gh123man

thijstriemstra, 2020-07-06:
> I can post a PR with the new resolution support. 

Pleaso do, we can discuss approach there.

gh123man, 2020-07-07:
Done. see: https://github.com/rm-hull/luma.oled/pull/291