rm-hull/luma.oled #292: Channel Already in use

nathan-stewart2, 2020-08-01:
I had this working, reinstalled OS after switching from Pizero to Pi3.  I can't seem to get the luma.spi stuff working. I have spi loaded (dtparam=spi=on,  modules spidev and spi_bcm2835 loaded, /dev/spidev0.0  exists.  If I load the RPi.GPIO module I can set/test the CS0 line, but with or without it, luma.core.interface.serial.spi can't seem to open spi0.0. Here's what happens when I try to talk to spi.

```
from luma.core.interface.serial import spi
serial = spi(device=0, port=0)
oled = ssd1322(serial, width=256, height=64, rotate=2)
```

/usr/local/lib/python3.7/dist-packages/luma/core/interface/serial.py:185: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(pin, self._gpio.OUT)
Traceback (most recent call last):
  File "./oled.py", line 18, in <module>
    serial = spi(device=0, port=0)
  File "/usr/local/lib/python3.7/dist-packages/luma/core/interface/serial.py", line 285, in __init__
    self._spi.cshigh = cs_high
SystemError: error return without exception set

I don't see any process being launched, however lsof shows that root has a spi0 process open (spawned by kthreadd). My user is in the spi group and it doesn't matter if I run the script as root - I still get the error. Any ideas what I'm doing wrong?

rm-hull, 2020-08-01:
See https://github.com/rm-hull/luma.led_matrix/issues/225

Try `sudo pip3 install --upgrade --force-reinstall luma.oled luma.core spidev`

nathan-stewart2, 2020-08-01:
Tried with same results. I did see this however:
luma-core 1.14.1 has requirement spidev<=3.4; platform_system == "Linux", but you'll have spidev 3.5 which is incompatible.

I downgraded which cleared the error, but it's still not working (when it was prior)

justanotherjohn, 2020-08-03:
There are two more of us with the exact same issue. We think it might have occurred with the last Raspian update.

george-howell, 2020-08-03:
There is now a third with the exact same issue. Stopped working after a updating Raspian.

MaryAlphaXray, 2020-08-05:
I to am having the same problem "SystemError: error return without exception set".  I tried the suggestions here but it still doesn't work. I have eight Pi's that have clock displays that no longer work as of an upgrade I did a week or so ago.

There is a lot more than just a handful of people with this problem.

nathan-stewart2, 2020-08-06:
Does anyone have a last known good data point? Here's a bit more on who has the spi device open:
sudo lsof |grep spi
spi0       179                           root  cwd       DIR      179,2     4096          2 /
spi0       179                           root  rtd       DIR      179,2     4096          2 /
spi0       179                           root  txt   unknown                                /proc/179/exe

However /proc/179/cmdline is empty and while there is a /dev/spi0 device file, there doesn't appear to be any sort of executable called spi0 to run. I also see no spi related entries in /var/log/apt (but that only means it didn't happen post imaging.) 
top - 00:59:47 up 22:39,  1 user,  load average: 0.00, 0.00, 0.00
Tasks:   1 total,   0 running,   1 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.1 us,  0.0 sy,  0.0 ni, 99.9 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :    926.8 total,    519.5 free,     41.4 used,    365.9 buff/cache
MiB Swap:    100.0 total,    100.0 free,      0.0 used.    814.6 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                         
  179 root      20   0       0      0      0 S   0.0   0.0   0:00.00 spi0

rm-hull, 2020-08-06:
@nathan-stewart2 

> Tried with same results. I did see this however:
> luma-core 1.14.1 has requirement spidev<=3.4; platform_system == "Linux", but you'll have spidev 3.5 which is incompatible.

Can you try this instead:

```console
$ sudo pip3 install --upgrade --upgrade luma.oled luma.core
```

MaryAlphaXray, 2020-08-06:
I tried "sudo pip3 install --upgrade --upgrade luma.oled luma.core" andI get "SystemError: error return without exception set"
That doesn't fix it.
Max

rm-hull, 2020-08-09:
What version of spidev do you have installed?

Try: `pip3 list | grep spidev` to see the version.

If it is 3.5 please try to downgrade it with `pip3 install spidev==3.4 --force-reinstall`

nathan-stewart2, 2020-08-09:
In my 2nd post I mentioned that downgrading makes the error go away, however it still doesn't actually work.

rm-hull, 2020-08-11:
> however it still doesn't actually work

Can you be more descriptive?
How have you got the display wired up?
If you run the main luma.examples demo, what do you see on the screen? What command-line args did you use?

MaryAlphaXray, 2020-08-11:
I have tried all the suggestions here which are,
sudo pip3 install --upgrade luma.oled luma.coresudo pip3 install --upgrade --upgrade luma.oled luma.corepip3 install spidev==3.4 --force-reinstallpip3 list | grep spidev  which shows 3.4
I get "SystemError: error return without exception set" whenever I try to run your python clock program. This had been working without any problems prior to doing an update/upgrade a few weeks ago. 
Please let me know if there is anything you would like me to try.
Max

rm-hull, 2020-08-11:
@MaryAlphaXray 
> try to run your python clock program

Please give the exact command you are using

MaryAlphaXray, 2020-08-11:
Either ./clock_7seg.py or python clock_7seg.py gives the same error.
Traceback (most recent call last):  File "clock_7seg.py", line 63, in <module>    main()  File "clock_7seg.py", line 45, in main    serial = spi(port=0, device=0, gpio=noop())  File "/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py", line 281, in __init__    self._spi.cshigh = cs_highSystemError: error return without exception set
Max

rm-hull, 2020-08-11:
@MaryAlphaXray 
Notice this: `/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py` ... python 2.7

I imagine if you did `pip list | grep spidev` it will show spidev version 3.5 is installed under python 2.7

Try: `python3 clock_7seg.py`

thijstriemstra, 2020-08-11:
I'll pin this issue since it comes up a lot lately.

MaryAlphaXray, 2020-08-12:
Running pip list | grep spidev shows 3.4.
When I run "python3 clock_7seg.py" I get
python3 clock_7seg.py  File "clock_7seg.py", line 48    seg.device.contrast(0x00)                            ^TabError: inconsistent use of tabs and spaces in indentation
I don't know enough about python to know how to fix this either.
Thanks, Max

rm-hull, 2020-08-12:
@MaryAlphaXray Where did you get the `clock_7seg.py` file? It is not one I recognise from the luma.led-matrix nor luma.examples repos.

Have a look at this to fix the tabs vs. spaces issue: https://stackoverflow.com/questions/5685406/inconsistent-use-of-tabs-and-spaces-in-indentation. I would recommend using autopep8 as per the second answer to fix your issue.

MaryAlphaXray, 2020-08-12:
The program is one of yours that has been renamed, sorry can't seem to find the original at the moment. It showed the date / time on a 8 character 7 segment display. I removed the date portion to show time only.
After I switched to using python3 and cleaned up the indention's it is working again, thanks. 
Have you tried all of your example programs to see if they would run under python 3 ?
Thanks again, Max

nathan-stewart2, 2020-08-15:
Ok, I can confirm it's working on Ubuntu 18.04, spidev 3.4,

MaryAlphaXray, 2020-08-17:
Just so you know, I've also had to install libtiff5 and libopenjp2-7 on some of my Pi's to fix other errors when switching from Python to Python3. Even though spidev showed 3.4 I still had to run
pip3 install spidev==3.4 --force-reinstall

on some but not all of my Pi's.
Max

guyjeangilles, 2020-08-27:
+1 seeing this error but I'm trying to get a display up and running for the first time. I've gone throw the installation as described in the wiki and tried the commands in this thread yet no dice

justanotherjohn, 2020-08-27:
When I did the down grade to      pip3 install spidev==3.4 --force-reinstall
that worked for me.

guyjeangilles, 2020-08-27:
Strange, `pip3 list | grep spidev` showed I had spidev 3.4 but `pip3 install spidev==3.4 --force-reinstall` removed the error. Still troubling the display but that may be a unrelated problem

closed 2020-10-11 by rm-hull