rm-hull/luma.oled #294: Support for Winstar WS0010 driver

GRA0007, 2020-08-24:
I recently purchased a [Winstar WEG010016A display](https://www.winstar.com.tw/products/oled-module/graphic-oled-display/weg010016a.html), incorrectly thinking that there would be ample documentation on how to use it online. I've spent several hours trawling through the internet to find if others have used it before with a raspberry pi and I managed to find a few references to different winstar _character_ displays being used, but none of this model, and none even of graphical winstar oled screens controlled by the WS0010 universal driver.

I did manage to find [this pull request](https://github.com/rm-hull/luma.core/pull/180) from 14 days ago which seems to add support for Winstar WEG displays to the luma.core, but as far as I can see (I'm quite a novice to SPI and displays in general), that support hasn't yet been added to the luma.oled project.

I have used this library before for a SSD1322 oled display, but for my current project I needed a red on black oled, and the above model is the only one I could find.

My question is, am I able to use the luma.oled library with my Winstar WEG display, and if not, it would be great if support could be added. I'm aware I sound a bit arrogant but as much as I'd like to contribute myself, I am really not sure where to even begin; the datasheets for these displays send my head spinning :/

thijstriemstra, 2020-08-24:
ping @dhrone

dhrone, 2020-08-24:
Actually, I'm pretty close to being able to request a pull.  I've allowed perfection to be the enemy of progress so got distracted by a re-write of the font handler I've created to embed into the display driver.   Shouldn't be much longer.

rm-hull, 2020-08-24:
@GRA0007 A really valuable contribution would be to test the new driver @dhrone is writing!

thijstriemstra, 2020-08-24:
@dhrone you expect that particular screen will work with your pr?

dhrone, 2020-08-24:
Yes.  I'm very familiar with that device and the code will definitely work on it.

GRA0007, 2020-08-25:
> @GRA0007 A really valuable contribution would be to test the new driver @dhrone is writing!

Now that I can do!

dhrone, 2020-08-25:
I just requested a pull for the core components of the drivers.  So, we're very close to having the hd44780 and ws0010 drivers available.

PR #295 (2020-09-01): Adding ws0010 and winstar_weh drivers
Added ws0010 driver (and related winstar_weh) driver.  Should resolve #294
closed 2020-09-01 by thijstriemstra