rm-hull/luma.oled #311: Multiple spi displays -> need help

Maschine2501, 2020-11-04:
Hi!
I want to use more than one display on one spi line.
I researched this, and came to the conclusion, that only the cs/ce pin must be different.
So I wired two ssd1351 displays up and connected them as followed (BCM Numbers)

I guessed that I need to initialize the seccond display with [gpio_cs_spi](https://github.com/rm-hull/luma.core/blob/bc788fd0161884765ddef60100030a474866a300/luma/core/interface/serial.py#L331)...

#### 1st Display:
Mosi -> 10
SCLK -> 11
RES ->    25
DC ->     24
CS ->      8

#### 2nd Display:
Mosi -> 10
SCLK -> 11
RES ->    25
DC ->     24
CS ->      7

#### Results:
Only on Display 2 (cs - 7) i get a picture.
If i wire it up so that dc, reset and cs are individual for each display and only Mosi and Sclk are shared, i get the same results.

#### Type of Raspberry Pi
Raspberry Pi 4 / 4GB
spi enabled
latest buster lite

#### Linux Kernel version
Linux raspberrypi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux

#### Expected behaviour
Both displays will display a picture (in the demo code below it is a text...) undependend to each other.
Goal is that i can display something else on each display with minimal use of other GPIO's.

#### Actual behaviour
Only one Display shows something.
Allready testet the displays, both work just fine in "mono" mode without the other.
I even tried an ssd1322 + ssd1351 same result.

#### The Code: 
( I've left the stuff I tried in the code, but commented)
```
#!/usr/bin/python3
import os
import sys
import time
from PIL import Image
from PIL import ImageDraw
from PIL import ImageFont
from luma.core.interface.serial import spi
#from luma.oled.device import ssd1322
from luma.core.interface.serial import gpio_cs_spi
from luma.oled.device import ssd1351

width1322 = 128
height1322 = 128
width1351 = 128
height1351 = 128

font = ImageFont.load_default()

interface22 = gpio_cs_spi(device=0, port=0, gpio_DC=14, gpio_RST=15, gpio_CS=7)
#interface22 = gpio_cs_spi(device=0, port=0, gpio_CS=8)
#interface22 = spi(device=0, port=0)
#oled22 = ssd1322(interface22) 
oled22 = ssd1351(interface22) 
oled22.clear()
interface51 = gpio_cs_spi(device=1, port=0, gpio_CS=7)
oled51 = ssd1351(interface51) 
oled51.clear()

image22 = Image.new('RGB', (width1322, height1322))
image51 = Image.new('RGB', (width1351, height1351))

draw22 = ImageDraw.Draw(image22)
draw51 = ImageDraw.Draw(image51)

def ssd1322picture():
    draw22.text((0, 20), "SSD1322", font=font, fill='white')
    oled22.display(image22)

def ssd1351picture():
    draw51.text((0, 64), "SSD1351", font=font, fill=(0, 255, 0))
    oled51.display(image51)

while True:
    ssd1322picture()
    ssd1351picture()
    time.sleep(10.0)
    oled22.clear()
    oled51.clear()
```

#### Can anyone help me with this topic? Is it even possible?

Maschine2501, 2020-11-04:
Hey Guys, 
sorry, but I have absolutely no Idea how to do this with a pull request, so I made a little Instruction here:

## The wiring:
### 1st Display:                    Raspberry Pi                            2nd Display:
Mosi/din ---------->     BCM-10 / Physical 19    <-----Mosi/din
SCLK/clk  ---------->     BCM-11 / Physical 23    <-----SCLK/clk
RES ---------------->     BCM-25 / Physical 22    <-----RES
DC ---------------->      BCM-24 / Physical 18    <-----DC
CS ----------------->      BCM-8   / Physical 24 
                                       BCM-7   / Physical 26    <-----CS

## Example Code:
```
#!/usr/bin/python3
import os
import sys
import time

#Imports for PIL (needed to "draw" pictures)
from PIL import Image
from PIL import ImageDraw
from PIL import ImageFont

#Imports from Luma (needed to display the pictures on the Oled)
from luma.core.interface.serial import spi
from luma.oled.device import ssd1322
from luma.core.interface.serial import gpio_cs_spi
from luma.oled.device import ssd1351

#Definition for Display Size in Pixel
width1st = 256
height1st = 64
width2nd = 128
height2nd = 128

#Definition for font
font = ImageFont.load_default()

#Initialisation of the Interface(s)
interface1st = spi(device=0, port=0)      #, gpio_RST=14)  <- can be used if you want seperate RST, in my case it was not needed
interface2nd = gpio_cs_spi(device=1, port=0, gpio_CS=7)

#Initialisation of the Screens
oled1st = ssd1322(interface1st) 
oled2nd = ssd1351(interface2nd) 

#Clear the Screens
oled1st.clear()
oled2nd.clear()

#Initialize Images in PIL for each Screen
image1st = Image.new('RGB', (width1st, height1st))
image2nd = Image.new('RGB', (width2nd, height2nd))

#Initialize the "draw" function of PIL for each image
draw1st = ImageDraw.Draw(image1st)
draw2nd = ImageDraw.Draw(image2nd)

#Make some definitions on what to draw:
def ssd1322picture():
    draw1st.text((0, 20), "SSD1322", font=font, fill='white')
    oled1st.display(image1st)

def ssd1351picture():
    draw2nd.text((0, 64), "SSD1351", font=font, fill=(0, 255, 0))
    oled2nd.display(image2nd)

#While Loop
while True:
    ssd1322picture()
    ssd1351picture()
```

rm-hull, 2020-11-05:
Thanks @Maschine2501, I’ll add it to the python usage docs

Maschine2501, 2020-11-15:
Hi guys! In the last days i noticed a strange behaviour. In my testfile i wrote one line (everything else stayed black), so i did not notice it before:
On the second screen, the previous content will remain until it is "overwritten" with color.
[image]

example: I had some text drawn on the complete screen, and afterwards i shutdown my pi.
Now when i want to display sometthing, only the parts of the picture where the new color != black is displayed, the rest only displays the previous Picture (even after loong powerloss and multiple reboots).
I tried to figure out the Answers of you, but  every configuration i tested , resulted in this case or that both displays show  the same picture.
I tested:
(MOSI, DC, SCLK parralell...)
Same CS, different RST
Same RST, differnet CS
Different CS and RS
Different CS, second screen no RST.

Maybe you can provide me with an example on how to wire and how to init?
Please help me with this, I'm kinda stuck....

Thank you

thijstriemstra, 2020-11-15:
@Maschine2501 what version of luma.oled and luma.core?

Maschine2501, 2020-11-15:
How do i check that?

thijstriemstra, 2020-11-15:
`python -c "from luma.oled import __version__; print(__version__);"`

for luma.core use same cmd but substitute core for oled

thijstriemstra, 2020-11-15:
ok that looks like the latest versions, thanks.

Maschine2501, 2020-11-15:
I need to mention: in the picture above,  the red circle shows the "new" content, simple draw.text "oled Nr. 2".
And i use two spi1322, one 3.12" and one 5.5".

Maschine2501, 2020-11-15:
Here are some more informations:
```
#!/usr/bin/python3
import os
import sys
import time
from time import *
from PIL import Image as Image1
from PIL import Image as Image2
from PIL import ImageDraw as ImageDraw1
from PIL import ImageDraw as ImageDraw2
from PIL import ImageFont
from luma.core.interface.serial import spi
from luma.oled.device import ssd1322
from luma.core.interface.serial import gpio_cs_spi

width = 256
height = 64

varcanc = True
secvar = oledPause2StopTime = 2

font = ImageFont.load_default()

interface1 = spi(device=0, port=0) #, gpio_RST=25, gpio_CS=8) #, gpio_RST=14) #spi(device=0, port=0) #, gpio_RST=14)
interface2 = gpio_cs_spi(device=1, port=0, gpio_CS=7)  #gpio_RST=14, , gpio_RST=14) #) #, gpio_DC=14) #, gpio_RST=25) gpio_cs_spi(device=1, port=0, gpio_CS=7, reset_hold_time=0.2)
oled1 = ssd1322(interface1) 
oled2 = ssd1322(interface2) 
oled1.clear()
oled2.clear()

def pictureNR1():
    image1 = Image1.new('RGB', (width, height))
    draw1 = ImageDraw1.Draw(image1)
    draw1.text((40, 20), "Oled Nummer 1", font=font, fill='white')
    oled1.display(image1)

def pictureNR2():
    oled2.clear()
    image2 = Image2.new('RGB', (width, height))
    draw2 = ImageDraw2.Draw(image2)
    draw2.text((40, 20), "Oled Nummer2", font=font, fill=(0, 255, 0))
    print('1112222')
    oled2.display(image2)
    

def pictureNR12():
    image1 = Image1.new('RGB', (width, height))
    draw1 = ImageDraw1.Draw(image1)
    draw1.rectangle((0, 0, 263, 63), fill='white', outline='white')
    oled1.display(image1)

def pictureNR22():
    oled2.clear()
    image2 = Image2.new('RGB', (width, height))
    draw2 = ImageDraw2.Draw(image2)
    draw2.rectangle((0, 0, 263, 63), fill='white', outline='white')
    print('2222222')
    oled2.display(image2)
    

while True:
    if varcanc == True:
        print('1tes if')
        secvar = int(round(time()))
        varcanc = False
        pictureNR1()
        pictureNR2()
    if int(round(time())) - secvar > oledPause2StopTime:
        print('2tes if')
        pictureNR12()
        pictureNR22()
        sleep(2.0)
        varcanc = True
        secvar = 0.0
```
I even tried to import PIL with two seperate names... You see the behaviour here:

[image]

PS: All, exept CS, is parallel.

rm-hull, 2020-11-15:
Can you try running any of the luma.example demo on your SSD1322's - particularly `demo.py` and maybe `jetset_willy.py` to see if they  exhibit any weird artifacts.

Also, to rule out the diff-to-previous changes that happened in luma.core 2.0.1, try changing your code as follows:

```python
#Initialisation of the Screens
from luma.core.framebuffer import full_frame
oled1 = ssd1322(interface1, framebuffer=full_frame()) 
oled2 = ssd1322(interface2, framebuffer=full_frame()) 
```
Let me know if that changes anything.

rm-hull, 2020-11-15:
Actually, I just thought - you will need to make sure each device has its own RESET line, as per @CRImier noted here: https://github.com/rm-hull/luma.oled/issues/311#issuecomment-721347009

Maschine2501, 2020-11-15:
@rm-hull : Thank you so much! TThatt was it!! 
I now use Mosi, Sclk and DC Parallel. Each Display has now his own RST and CS:
```
interface = spi(device=0, port=0) 
interface2 = gpio_cs_spi(device=1, port=0, gpio_CS=7, gpio_RST=14)
oled = ssd1322(interface, rotate=oledrotation, framebuffer=full_frame())
oled2 = ssd1322(interface2, framebuffer=full_frame())
```
I tested this wiring before, without succes, so the Framebuffer did it!
Thank you!!!!!!!! <3

closed 2020-11-15 by Maschine2501

rm-hull, 2020-11-15:
Errr, so could you temporarily remove `framebuffer=full_frame()` from both. We just need to establish if it was the gpio reset lines that caused this or whether there is a bug in the diff-to-previous framebuffer class

Maschine2501, 2020-11-15:
> 
> 
> Errr, so could you temporarily remove `framebuffer=full_frame()` from both. We just need to establish if it was the gpio reset lines that caused this or whether there is a bug in the diff-to-previous framebuffer class

Hi, I just removed  `framebuffer=full_frame()` from both displays -> same error as before.
If I put `framebuffer=full_frame()` back in place, both displays work like charm.
I even double checked with another pi and two other ssd1322 displays.

I left the cabling as followed:
Wired like this:
Display1--------------RPi-----------Display2
Mosi---------------Mosi(10)-----------Mosi
SCLK---------------SCLK(11)----------SCLK
DC------------------DC(24)-------------DC
CS-------------------CS0(8)xxxxxxxxxxxxxxxxx
RST-----------------RST(25)xxxxxxxxxxxxxxxxx
xxxxxxxxxxxxxxxxxxxxxCS1(7)--------------CS
xxxxxxxxxxxxxxxxxxxxxRST(14)------------RST

Code:
```
#!/usr/bin/python3
import os
import sys
import time
from time import *
from PIL import Image as Image1
from PIL import Image as Image2
from PIL import ImageDraw as ImageDraw1
from PIL import ImageDraw as ImageDraw2
from PIL import ImageFont
from luma.core.framebuffer import full_frame
from luma.core.interface.serial import spi
from luma.oled.device import ssd1322
from luma.core.interface.serial import gpio_cs_spi

width = 256
height = 64

varcanc = True
secvar = oledPause2StopTime = 2

font = ImageFont.load_default()

interface1 = spi(device=0, port=0)
interface2 = gpio_cs_spi(device=1, port=0, gpio_CS=7, gpio_RST=14) 
oled1 = ssd1322(interface1) #, framebuffer=full_frame()) 
oled2 = ssd1322(interface2) #, framebuffer=full_frame()) 
oled1.clear()
oled2.clear()

image1 = Image1.new('RGB', (width, height))
draw1 = ImageDraw1.Draw(image1)

image2 = Image2.new('RGB', (width, height))
draw2 = ImageDraw2.Draw(image2)

def pictureNR1():
    image1 = Image1.new('RGB', (width, height))
    draw1 = ImageDraw1.Draw(image1)
    draw1.text((40, 20), "Oled Nummer 1", font=font, fill='white')
    oled1.display(image1)

def pictureNR2():
    oled2.clear()
    image2 = Image2.new('RGB', (width, height))
    draw2 = ImageDraw2.Draw(image2)
    draw2.text((40, 20), "Oled Nummer2", font=font, fill=(0, 255, 0))
    print('1112222')
    oled2.display(image2)

def pictureNR12():
    image1 = Image1.new('RGB', (width, height))
    draw1 = ImageDraw1.Draw(image1)
    draw1.rectangle((0, 0, 263, 63), fill='white', outline='white')
    oled1.display(image1)

def pictureNR22():
    oled2.clear()
    image2 = Image2.new('RGB', (width, height))
    draw2 = ImageDraw2.Draw(image2)
    draw2.rectangle((0, 0, 263, 63), fill='white', outline='white')
    print('2222222')
    oled2.display(image2)

while True:
    if varcanc == True:
        print('1tes if')
        secvar = int(round(time()))
        varcanc = False
        pictureNR1()
        pictureNR2()
    if int(round(time())) - secvar > oledPause2StopTime:
        print('2tes if')
        pictureNR12()
        pictureNR22()
        sleep(2.0)
        varcanc = True
        secvar = 0.0
```

Hope that helps ;)

rm-hull, 2020-11-15:
Ok, good to know, .... so we have uncovered a potential bug in how the diff-to-previous framebuffer implementation interacts with ssd1322. Out of curiosity, were you able to check the `demo.py` example to see if that glitched?

Maschine2501, 2020-11-15:
Hey, I'll test the demo file(s) tommorow... i'm off the pc now... sorry

rm-hull, 2020-11-15:
Ah, I think I know what the problem is... basically this: https://docs.python-guide.org/writing/gotchas/

Your two devices are sharing the same `diff-to-previous` instance because of the default parameter.

If you change your python code to:
```python
oled1 = ssd1322(interface1) #, framebuffer=diff_to_previous()) 
oled2 = ssd1322(interface2) #, framebuffer= =diff_to_previous())  
```
Then I believe this would work correctly, because they are two separate instances.

I will issue a fix for this shortly.
Many thanks for your help reporting this.