rm-hull/luma.oled #312: winstar eh1602a SPI interface

bernerha, 2020-11-09:
#### Type of Raspberry Pi
> Pi 3 Model B+

#### Linux Kernel version
> Linux moode 5.4.72-v7+ #1356 SMP Thu Oct 22 13:56:54 BST 2020 armv7l GNU/Linux

#### Expected behaviour
> I would like to render text on the display. I already tested the display with an arduino with spi and parallel interface and both works.
Here ist the arduino code for the spi interface
```
#define SPIPORT (PORTB)
#define SPITOGGLE (PINB)
// PB5 (0x20) is SCK  (output) green  OLED pin 12
#define SPI_SCK_PIN (13)
#define SCK_MASK (0x20)
#define CLR_SCK (PORTB &= ~(SCK_MASK))
#define SET_SCK (PORTB |=  (SCK_MASK))
// PB4 (0x10) is MISO (input)  blue   OLED pin 13
//(reference only, it is an input)
#define SPI_MISO_PIN (12)
#define MISO_MASK (0x10)
#define CLR_MISO (PORTB &= ~(MISO_MASK))
#define SET_MISO (PORTB |=  (MISO_MASK))
// PB3 (0x08) is MOSI (output) violet OLED pin 14
#define SPI_MOSI_PIN (11)
#define MOSI_MASK (0x08)
#define CLR_MOSI (PORTB &= ~(MOSI_MASK))
#define SET_MOSI (PORTB |=  (MOSI_MASK))
// DB2 (0x04) is SS   (output) gray   OLED pin 16
#define SPI_SS_PIN (10)
#define SS_MASK (0x04)
#define CLR_SS  (PORTB &= ~(SS_MASK))
#define SET_SS  (PORTB |=  (SS_MASK))

#define DATA (1)
#define COMMAND (0)
//===========================================================================
void write_to_OLED_SPI(uint8_t destination,uint8_t data)
  {
  //Bits sent in this order:
  //  data(1)/command(0) flag
  //  read(1)/write(0) flag
  //  data.7 (msb)
  //  data.6
  //  data.5
  //  data.4
  //  data.3
  //  data.2
  //  data.1
  //  data.0 (lsb)

#if(1)
  //Pretty fast software SPI 10-bit transfer code
  //Several ideas taken from the fastest non-assembly implementation at:
  //http://nerdralph.blogspot.com/2015/03/fastest-avr-software-spi-in-west.html
  //
  // The SS pin is low for ~3.8uS, the clock frequency is ~ 2.6MHz
  // 47x faster than above

  //Pre-calculate the value that will drive clk and data low in one cycle.
  //(Note that this is not interrupt safe if an ISR is writing to the same port)
  register uint8_t
    force_clk_and_data_low;

  //Select the chip
  CLR_SS;

  //Pre-calculate the value that will drive clk and data low in one
  //operation.
  force_clk_and_data_low = (SPIPORT & ~(MOSI_MASK | SCK_MASK) );
[... 282 lines cut ...]
  }
//=====================================

```

#### Actual behaviour
> Nothing happens with following code:
```
from time import sleep
from luma.core.interface.serial import spi
from luma.oled.device import ws0010

p = spi(device=0, port=0)
d = ws0010(p, selected_font='FT01')
sleep(2)
d.text = 'HELLO'
sleep(10)
```

I use the 3-wire SPI and connected through a logic level converter SCK, MISO, MOSI, SS (CE0), VSS, VDD.
Maybe I missing something like the Data/Command Pin which I want to use inline instead of a extra pin.

My goal is to use the pydPiper lib somehow. The pydPiper lib doesn't support SPI yet but I saw dhrone pushing some code here.

Regards

dhrone, 2020-11-09:
Have you tried testing your weh001602a with Luma.Oled using the parallel style interface?  

I do not have any documentation on how to drive a Winstar weh001602a using SPI.  If you could point me in the direction of some it will help me to determine if there is something I can recommend.  Also, it would be helpful to know how you had the display wired to your arduino when running the above code.

bernerha, 2020-11-10:
I also tried the parallel interface with luma.oled but I got nothing on the display. I tried it with 3V and 5V and four data wires (d4-d7) like in the examples but without a logic level converter because mine only has 4 connections.

The connection to the arduino with SPI:
Arduino <-> Display
13 <-> 12 (SCK)
12 <-> 13 (MISO)
11 <-> 14 (MOSI)
10 <-> 16 (SS)

The connection in parallel mode on the raspberry was like in the luma.oled example https://luma-oled.readthedocs.io/en/latest/hardware.html#parallel

I also tried the pydPiper docker image with no results. The fonts took a really long time to load and the container did nothing else. I already used a hd44780 display which worked without any problems.

dhrone, 2020-11-10:
That is odd.  I note that your issue title says you are using the eh1602a.  I had thought that was a typo and that you were actually using a weh1602a.  Those displays do not use pins 15 and 16.  Could you confirm the display you are using?  For what it's worth, I am also unaware that the weh1602a supports an SPI interface but would happy to be corrected.

bernerha, 2020-11-11:
The original weh001602a has a SPI interface, see here https://www.winstar.com.tw/products/oled-module/oled-character-display/weh001602a.html
But it's unusual to use I think. It took me some time to find the working arduino example :)
It's not a original winstar display but it looks the same and uses the same pinout. The pins 15 and 16 probably won't get used in parallel mode.

dhrone, 2020-11-11:
I see that it is available as an option but there is no description as to how it is implemented on the device making it very difficult to determine how to guide you.  SPI is a relatively loose standard so having clear documentation on how the SPI for your display is implemented is important for debugging your problem.

It is possible that your SPI implementation is using a different mode assumption than the spidev based implementation that Luma uses.  You may want to try using different SPI modes.  There are four possible values (0, 1, 2, 3) which can be specified using the spi_mode argument of class spi.

`p = spi(device=0, port=0, spi_mode=n)`
where _n_ is an integer in the range 0-3

That said, I know that the weh001602a works correctly using the 4 bit parallel wiring provided in the documentation.  Perhaps we should focus on getting your display set up using that method.

bernerha, 2020-11-11:
I tried the spi_mode option and used the reset_release_time and reset_hold_time with one second with no result.
Ok let's try the parallel mode. I will solder the jumpers later and attach an image with the wiring.

bernerha, 2020-11-11:
Okay I resoldered the display and verified it's working with the arduino.
I connected the pins exactly like here https://luma-oled.readthedocs.io/en/latest/hardware.html#parallel

Arduino setup as picture.
[image]

bernerha, 2020-11-12:
Solved it!. Rearranged the RS and E pin away from GPIO7/8 which are the SPI pins (SPI is deactivated) and it works now.
Startes rearranging because I checked my soundcard pins usage...

closed 2020-11-12 by bernerha

bernerha, 2020-11-12:
fyi:
JustBoom DAC HAT
Functional pin occupancy (GPIO numbers):
RS -> 12
E -> 5
D4 -> 20
D5 -> 16
D6 -> 13
D7 -> 6
GND -> GND
VDD -> 3.3V

dhrone, 2020-11-12:
Excellent news.  Interesting about GPIO pins 7/8.  While their use shouldn't matter if SPI is not enabled perhaps the documentation should be updated to avoid pins that have special purposes.  I would still be interested in any materials that you have discovered that describe the SPI interface capabilities of the weh001602a.