rm-hull/luma.oled #314: SSD1306 controller with 128x32 display image issue.

DJ-Dingo, 2020-11-19:
I been making a plugin where i use various oled screens and It's all working fine, 
but when i use SSD1306 with an 128x32 it wont display the 128x32 images, it just go black screen.
I have trying divice.height or force it 32. but still wont display them.

But if i use an 128x64 images it works fine, "but shrunk the images of course". 
(with is fine if an 128x32 image is not present in the folder) but want it to display the 128x32 image.

luma.oled 3.8.1 - Tests working fine.
$ python3 examples/demo.py -d ssd1306  --i2c-port=1 --width=128 --height=32

And if i use the old adafruit_ssd1306 driver, it is all working fine too.
It can show both the 128x32 and 128x64 images (x64 shrunk the image of course)

It looks like it can't detect that it is a 128x32 OLED automatic and read it as and 128x64 device.

In the class ssd1306(device):  
def __init__  
the height is set to height=64

What can i do, or is it something in the luma.oled 3.8.1
All my 128x64 are working fine.

thijstriemstra, 2020-11-19:
what code are you using? try `device = ssd1306(serial, width=128, height=32)`

DJ-Dingo, 2020-11-19:
thx for the fast reply :-) 

I switch between various screens from a menu, so can 't hardcode it
I allready tried to test with device = ssd1306(serial, width=128, height=32) not working. (but working if i import and use adafruit_ssd1306)

Well it is kind of working, but i am using various screens, so i have to use 
device = ssd1306(serial, rotate=0) and get the device.width, device.height in 
image = Image.new(mode=1, (device.width, device.height))
(i do some more tests)

it fails in my line with the "device.display(image)"
can i set device.width, device.height in the     device.display(image) ??

File "/home/pi/.local/lib/python3.7/site-packages/luma/oled/device/greyscale.py", line 121, in display
    assert(image.mode == self.mode)

My 128c64 displays are working fine, so thats why i don't get it.

DJ-Dingo, 2020-11-20:
Don't no whatś wrong seems to work sometimes.
So close this for now.

closed 2020-11-20 by DJ-Dingo

rm-hull, 2020-11-20:
if it intermittently works,  then this points to a wiring or noise issue

DJ-Dingo, 2020-11-20:
@rm-hull 
yeah or wrong programing from my side, couse its working fine now. :-) 

Would it be possible with an auto-detect with what chip of OLED screen you are using ?

rm-hull, 2020-11-21:
> Would it be possible with an auto-detect with what chip of OLED screen you are using ?

That would be nice, but certainly for SPI devices, the comms is one-way only from the RPi to the display: you generally connect GPIO 10 as the data line - this is the MOSI (Master out, serial in). see https://pinout.xyz/pinout/pin19_gpio10. To be able to read data you would need to connect GPIO 9 (MISO), and every display I've ever seen never has had a data out to be able to connect that pin.

I2C supports bidirectional comms, so it may be possible for devices connected via that, but I don't ever recall seeing any documentation of any registers that can be poked to get a device id response. Happy to corrected on this if other folks know differently.

DJ-Dingo, 2020-11-22:
Thx for info. I mostly use I2c for now. 
I let you know if i find any info on that part :-)