rm-hull/luma.oled #315: Python 3 Luma can't be found

scargill, 2020-11-20:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

scargill, 2020-11-20:
Hi I'm having troublke getting LUMA started, I've used it msany times in the past.

I'm using a Raspberry Pi 4 - with Buster - fully up to date. I'm using sn SSD1306 display.

I have the simplest test page and I have Python 2.7 and 3.7

python crap.py
pi@rpi419buster:~:17:05[0]> python crap.py
Traceback (most recent call last):
  File "crap.py", line 1, in <module>
    from luma.core.interface.serial import i2c, spi, pcf8574
ImportError: No module named luma.core.interface.serial

trying python3

pi@rpi419buster:~:17:05[130]> python3 crap.py
Traceback (most recent call last):
  File "crap.py", line 1, in <module>
    from luma.core.interface.serial import i2c, spi, pcf8574
ModuleNotFoundError: No module named 'luma'
pi@rpi419buster:~:17:06[1]>

Python3 says no luma yet I used PIP3 show and it says yes?
pi@rpi419buster:~:17:06[130]> pip3 show luma.core
Name: luma.core
Version: 2.0.1
Summary: A component library to support SBC display drivers
Home-page: https://github.com/rm-hull/luma.core
Author: Richard Hull
Author-email: richard.hull@destructuring-bind.org
License: MIT
Location: /usr/local/lib/python3.7/site-packages
Requires: pillow, pyftdi, RPI.GPIO, smbus2, spidev, deprecated, cbor2
Required-by: luma.oled
pi@rpi419buster:~:17:07[0]>

Any ideas?

Pete

thijstriemstra, 2020-11-20:
python2 is not supported anymore, use python3.

rm-hull, 2020-11-20:
Hi Pete, you installed as per the instructions here: https://luma-oled.readthedocs.io/en/latest/software.html? 
Is there a package called **luma** installed? Try: `pip3 show luma` 
If so, please remove it, and reinstall **luma.oled**

scargill, 2020-11-20:
Thanks for the quick response - REALLY apperciated. Erm yes I installed (not from scratch as I had a working Adafruit version until it stopped working):

https://luma-oled.readthedocs.io/en/latest/software.html

And no there is no luma it seems. See below. Any pointers appreciated. I just want my SSD1306 working again.....

pi@rpi419buster:~:19:20[130]> pip3 show luma
WARNING: Package(s) not found: luma
pi@rpi419buster:~:19:20[1]>

scargill, 2020-11-20:
> python2 is not supported anymore, use python3.

I tried. No  joy, just replied...

scargill, 2020-11-20:
pip3 show luma.oled shows v 3.8.1  and pip3 show luma.core shows v2.0.1 - yet python3
 crap.py - returns - modulenotfounderror: no module named luma.

rm-hull, 2020-11-21:
If you run `ls -lR /usr/local/lib/python3.7/dist-packages/luma*`, could you paste the output here please

scargill, 2020-11-21:
Just checked Richard - there is /usr/local/lib/python3.7/dist-packages/ but nothing in it

in  /usr/local/lib/python3.7/site-packages there is a TON of stuff including luma, luma-core-2.01.dist-info and luma.oled-3.8.1.dist-info 

Does that help?

thijstriemstra, 2020-11-21:
@scargill try `ls -lR /usr/local/lib/python3.7/site-packages/luma*`

scargill, 2020-11-21:
pi@rpi419buster:~:12:38[0]> ls -lR /usr/local/lib/python3.7/site-packages/luma*
-rw-r--r-- 1 root staff  529 Nov 20 16:47 /usr/local/lib/python3.7/site-packages/luma.core-2.0.1-py3.8-nspkg.pth
-rw-r--r-- 1 root staff  529 Nov 20 16:47 /usr/local/lib/python3.7/site-packages/luma.oled-3.8.1-py3.8-nspkg.pth

/usr/local/lib/python3.7/site-packages/luma:
total 8
drwxr-sr-x 5 root staff 4096 Nov 20 16:47 core
drwxr-sr-x 4 root staff 4096 Nov 20 16:47 oled

/usr/local/lib/python3.7/site-packages/luma/core:
total 152
-rw-r--r-- 1 root staff  2974 Nov 20 16:47 ansi_color.py
-rw-r--r-- 1 root staff 24322 Nov 20 16:47 bitmap_font.py
-rw-r--r-- 1 root staff 16343 Nov 20 16:47 cmdline.py
-rw-r--r-- 1 root staff   348 Nov 20 16:47 const.py
-rw-r--r-- 1 root staff  9170 Nov 20 16:47 device.py
-rw-r--r-- 1 root staff   873 Nov 20 16:47 error.py
-rw-r--r-- 1 root staff  5184 Nov 20 16:47 framebuffer.py
-rw-r--r-- 1 root staff  5157 Nov 20 16:47 image_composition.py
-rw-r--r-- 1 root staff   134 Nov 20 16:47 __init__.py
drwxr-sr-x 3 root staff  4096 Nov 20 16:47 interface
drwxr-sr-x 3 root staff  4096 Nov 20 16:47 legacy
-rw-r--r-- 1 root staff  1131 Nov 20 16:47 lib.py
-rw-r--r-- 1 root staff  2559 Nov 20 16:47 mixin.py
drwxr-sr-x 2 root staff  4096 Nov 20 16:47 __pycache__
-rw-r--r-- 1 root staff  1727 Nov 20 16:47 render.py
-rw-r--r-- 1 root staff  8886 Nov 20 16:47 sprite_system.py
-rw-r--r-- 1 root staff  1312 Nov 20 16:47 threadpool.py
-rw-r--r-- 1 root staff  3920 Nov 20 16:47 util.py
-rw-r--r-- 1 root staff 21085 Nov 20 16:47 virtual.py

/usr/local/lib/python3.7/site-packages/luma/core/interface:
total 44
-rw-r--r-- 1 root staff   109 Nov 20 16:47 __init__.py
-rw-r--r-- 1 root staff  4147 Nov 20 16:47 parallel.py
drwxr-sr-x 2 root staff  4096 Nov 20 16:47 __pycache__
-rw-r--r-- 1 root staff 27439 Nov 20 16:47 serial.py

/usr/local/lib/python3.7/site-packages/luma/core/interface/__pycache__:
total 40
-rw-r--r-- 1 root staff   157 Nov 20 16:47 __init__.cpython-37.pyc
-rw-r--r-- 1 root staff  4172 Nov 20 16:47 parallel.cpython-37.pyc
-rw-r--r-- 1 root staff 27813 Nov 20 16:47 serial.cpython-37.pyc

/usr/local/lib/python3.7/site-packages/luma/core/legacy:
total 100
-rw-r--r-- 1 root staff 90426 Nov 20 16:47 font.py
-rw-r--r-- 1 root staff  3138 Nov 20 16:47 __init__.py
drwxr-sr-x 2 root staff  4096 Nov 20 16:47 __pycache__

/usr/local/lib/python3.7/site-packages/luma/core/legacy/__pycache__:
total 40
-rw-r--r-- 1 root staff 33554 Nov 20 16:47 font.cpython-37.pyc
-rw-r--r-- 1 root staff  3339 Nov 20 16:47 __init__.cpython-37.pyc

/usr/local/lib/python3.7/site-packages/luma/core/__pycache__:
total 144
-rw-r--r-- 1 root staff  2614 Nov 20 16:47 ansi_color.cpython-37.pyc
-rw-r--r-- 1 root staff 20963 Nov 20 16:47 bitmap_font.cpython-37.pyc
-rw-r--r-- 1 root staff 13730 Nov 20 16:47 cmdline.cpython-37.pyc
-rw-r--r-- 1 root staff   520 Nov 20 16:47 const.cpython-37.pyc
-rw-r--r-- 1 root staff 10388 Nov 20 16:47 device.cpython-37.pyc
-rw-r--r-- 1 root staff  1575 Nov 20 16:47 error.cpython-37.pyc
-rw-r--r-- 1 root staff  4795 Nov 20 16:47 framebuffer.cpython-37.pyc
-rw-r--r-- 1 root staff  6039 Nov 20 16:47 image_composition.cpython-37.pyc
-rw-r--r-- 1 root staff   168 Nov 20 16:47 __init__.cpython-37.pyc
-rw-r--r-- 1 root staff  1015 Nov 20 16:47 lib.cpython-37.pyc
-rw-r--r-- 1 root staff  2826 Nov 20 16:47 mixin.cpython-37.pyc
-rw-r--r-- 1 root staff  1798 Nov 20 16:47 render.cpython-37.pyc
-rw-r--r-- 1 root staff  8655 Nov 20 16:47 sprite_system.cpython-37.pyc
-rw-r--r-- 1 root staff  1690 Nov 20 16:47 threadpool.cpython-37.pyc
-rw-r--r-- 1 root staff  6319 Nov 20 16:47 util.cpython-37.pyc
-rw-r--r-- 1 root staff 21644 Nov 20 16:47 virtual.cpython-37.pyc

/usr/local/lib/python3.7/site-packages/luma/oled:
total 72
-rw-r--r-- 1 root staff 59895 Nov 20 16:47 const.py
drwxr-sr-x 3 root staff  4096 Nov 20 16:47 device
-rw-r--r-- 1 root staff   258 Nov 20 16:47 __init__.py
drwxr-sr-x 2 root staff  4096 Nov 20 16:47 __pycache__

/usr/local/lib/python3.7/site-packages/luma/oled/device:
total 64
-rw-r--r-- 1 root staff  3134 Nov 20 16:47 color.py
-rw-r--r-- 1 root staff  1139 Nov 20 16:47 framebuffer_mixin.py
-rw-r--r-- 1 root staff  4567 Nov 20 16:47 greyscale.py
-rw-r--r-- 1 root staff 41781 Nov 20 16:47 __init__.py
drwxr-sr-x 2 root staff  4096 Nov 20 16:47 __pycache__

/usr/local/lib/python3.7/site-packages/luma/oled/device/__pycache__:
total 52
-rw-r--r-- 1 root staff  3370 Nov 20 16:47 color.cpython-37.pyc
-rw-r--r-- 1 root staff  1253 Nov 20 16:47 framebuffer_mixin.cpython-37.pyc
-rw-r--r-- 1 root staff  4552 Nov 20 16:47 greyscale.cpython-37.pyc
-rw-r--r-- 1 root staff 35632 Nov 20 16:47 __init__.cpython-37.pyc

/usr/local/lib/python3.7/site-packages/luma/oled/__pycache__:
total 20
-rw-r--r-- 1 root staff 15799 Nov 20 16:47 const.cpython-37.pyc
-rw-r--r-- 1 root staff   301 Nov 20 16:47 __init__.cpython-37.pyc

/usr/local/lib/python3.7/site-packages/luma.core-2.0.1.dist-info:
total 52
-rw-r--r-- 1 root staff     4 Nov 20 16:47 INSTALLER
-rw-r--r-- 1 root staff  1123 Nov 20 16:47 LICENSE.rst
-rw-r--r-- 1 root staff 26483 Nov 20 16:47 METADATA
-rw-r--r-- 1 root staff     5 Nov 20 16:47 namespace_packages.txt
-rw-r--r-- 1 root staff  3452 Nov 20 16:47 RECORD
-rw-r--r-- 1 root staff     5 Nov 20 16:47 top_level.txt
-rw-r--r-- 1 root staff   110 Nov 20 16:47 WHEEL

/usr/local/lib/python3.7/site-packages/luma.oled-3.8.1.dist-info:
total 52
-rw-r--r-- 1 root staff     4 Nov 20 16:47 INSTALLER
-rw-r--r-- 1 root staff  1123 Nov 20 16:47 LICENSE.rst
-rw-r--r-- 1 root staff 25522 Nov 20 16:47 METADATA
-rw-r--r-- 1 root staff     5 Nov 20 16:47 namespace_packages.txt
-rw-r--r-- 1 root staff  1610 Nov 20 16:47 RECORD
-rw-r--r-- 1 root staff     0 Nov 20 16:47 REQUESTED
-rw-r--r-- 1 root staff     5 Nov 20 16:47 top_level.txt
-rw-r--r-- 1 root staff   110 Nov 20 16:47 WHEEL
pi@rpi419buster:~:13:34[0]>

scargill, 2020-11-21:
I even tried removing all references to luma anything from /usr/local/lib/python3.7/site-packages/
then followed the instructions here - https://luma-oled.readthedocs.io/en/latest/software.html

No further forward, "No module named 'luma' and /usr/local/lib/python3.7/dist-packages/ still remains empty

rm-hull, 2020-11-21:
What version of pip3 is installed? try `pip3 --version`

I just realized that the help page says to use pip rather than pip3 (now corrected), so perhaps you could try the following, and then copy the output and paste it here again?

```console
$ sudo -H pip3 install --upgrade --force-reinstall luma.oled
```

Lastly, what does the following display:
```console
$ python3 -c "import sys; print('\n'.join(sys.path))"
```

scargill, 2020-11-22:
pi@rpi419buster:~:09:34[0]> pip3 --version
pip 20.2.4 from /home/pi/.local/lib/python3.7/site-packages/pip (python 3.7)

Did the SUDO -H pip3 line above...
things happened - lots of uninstalls - 
...
Successfully uninstalled luma.oled-3.8.1
Successfully installed RPI.GPIO-0.7.0 cbor2-5.2.0 deprecated-1.2.10 luma.core-2.0.1 luma.oled-3.8.1 pillow-8.0.1 pyftdi-0.52.0 pyserial-3.4 pyusb-1.1.0 smbus2-0.3.0 spidev-3.5 wrapt-1.12.1

BUT  /usr/local/lib/python3.7/dist-packages remains empty

Finally...

pi@rpi419buster:~:09:39[130]> python3 -c "import sys; print('\n'.join(sys.path))"

/usr/lib/python37.zip
/usr/lib/python3.7
/usr/lib/python3.7/lib-dynload
/home/pi/.local/lib/python3.7/site-packages
/usr/local/lib/python3.7/dist-packages
/usr/lib/python3/dist-packages
pi@rpi419buster:~:09:40[0]>

Finally...

pi@rpi419buster:~:09:40[130]> python3 -c "import sys; print('\n'.join(sys.path))"

/usr/lib/python37.zip
/usr/lib/python3.7
/usr/lib/python3.7/lib-dynload
/home/pi/.local/lib/python3.7/site-packages
/usr/local/lib/python3.7/dist-packages
/usr/lib/python3/dist-packages
pi@rpi419buster:~:09:45[0]>

rm-hull, 2020-11-22:
Ok, I wonder if the cause is that pip3 has been installed locally (as evidenced from `pip 20.2.4 from /home/pi/.local/lib/python3.7/site-packages/pip (python 3.7)`). This may be the reason why your sys.path points to /home/pi/.local/lib/python3.7/site-packages rather than the one in /usr/local

In which case you could try installing the luma packages locally with (note no sudo here):

```console
$ pip3 install --upgrade --force-reinstall luma.oled
```

I think the tldr here is that python packaging is still a mess... :(

scargill, 2020-11-22:
> Ok, I wonder if the cause is that pip3 has been installed locally (as evidenced from `pip 20.2.4 from /home/pi/.local/lib/python3.7/site-packages/pip (python 3.7)`). This may be the reason why your sys.path points to /home/pi/.local/lib/python3.7/site-packages rather than the one in /usr/local
> 
> In which case you could try installing the luma packages locally with (note no sudo here):
> 
> ```
> $ pip3 install --upgrade --force-reinstall luma.oled
> ```
> 
> I think the tldr here is that python packaging is still a mess... :(

Thanks - in the meantime I scrfubbed my one file and went to Adafruit in case there were no answers... bad move. Now I've grabbed your examples - which have been updated since I last used them - I normally specidfy the display - ssd1306 using 128*32. It seems your examples use something called demo_opts which I apparently don;t have -I've run your latest suggestion, looked like it installed but apparently I need demo-opts to try anything. Any idea? Sorry.

scargill, 2020-11-22:
python3 welcome.py
Traceback (most recent call last):
  File "welcome.py", line 13, in <module>
    from demo_opts import get_device
ModuleNotFoundError: No module named 'demo_opts'
pi@rpi419buster:~:15:08[1]> pip3 show luma.core
Name: luma.core
Version: 2.0.1
Summary: A component library to support SBC display drivers
Home-page: https://github.com/rm-hull/luma.core
Author: Richard Hull
Author-email: richard.hull@destructuring-bind.org
License: MIT
Location: /home/pi/.local/lib/python3.7/site-packages
Requires: pillow, spidev, RPI.GPIO, cbor2, pyftdi, deprecated, smbus2
Required-by: luma.oled

rm-hull, 2020-11-22:
If you are targeting a specific device, just import the relevant class and use that. The demo_opt / get_device is just a helper in the examples repo (https://github.com/rm-hull/luma.examples/blob/master/examples/demo_opts.py) to allow the examples to run on a wide variety of devices by specifying command line args

scargill, 2020-11-22:
Going to try that now, I'm a Python novice so I just hope "importing the relevant class" (I know what classes are but only done basic Python) is as simple as you make it sound. Thanks for that.. I use SSD1306, eitheth the 32 or 64 vertical dots by 128 across, I2c, no reset etc.  :-)

rm-hull, 2020-11-22:
Replace https://github.com/rm-hull/luma.examples/blob/master/examples/sys_info.py#L111 with:

```python
from luma.core.interface.serial import i2c
from luma.oled.device import ssd1306
serial = i2c(port=1, address=0x3C)
device = ssd1306(serial, width=128, height=32)
```

scargill, 2020-11-22:
My hero - I will check that after lockdown-breaking trip out tomorow to get the car MOT'd (or ITV'd as we call it in Spain). When I get back lunchtime I'm putting that code in - cheers.

Pete

closed 2020-11-27 by rm-hull