rm-hull/luma.oled #316: Supporting 4 line Winstar WEH displays

dhrone, 2020-11-23:
It turns out there are some Winstar WEH displays that support more than 2 lines specifically the Winstar WEH002004.  While these displays do use the ws0010 controller, they do NOT support the graphics mode unlike the WEH001602 displays which do.  So the `luma.oled.display.ws0010` and the `luma.oled.display.winstar_weh` drivers will not work with them.  What they need is a driver that is largely the same as `luma.lcd.display.hd44780` but with the initialization logic of the ws0010.  This should support any ws0010 powered device that needs to operate in character mode.

So my question is, would you prefer to see the character-mode functionality added to `luma.oled.display.winstar_weh` or the creation of a new driver perhaps named `luma.oled.display.winstar_weh_character`? 

It is worth noting that I do not own a display to test this on.  However, there is someone requesting this functionality over in the pydPiper project.  I've already implemented a version of the needed driver within pydPiper and I can see if the user is willing to test the resulting luma driver once it is completed.

rm-hull, 2020-11-23:
Perhaps a new driver.. is there anything more specific we could use than `luma.oled.display.winstar_weh_character` ?

dhrone, 2020-11-23:
Well, there are only a few displays using the ws0010 controller that I think would require the character-based approach but it's hard to tell exactly which ones need it due to the lack of specific, model-by-model documentation on winstar's website.  According to the ws0010 documentation I have, the graphics addressing is limited to 100x16 pixels (e.g. 20 chars x 2 lines) so I'd assume that any WEH display that advertises larger values than that are stuck to character mode.

As for naming, I could create a master class that supports everything (perhaps a modification of the existing weh driver) with subclasses that use the exact model number of the display with appropriate defaults set for the initialization parameters.