rm-hull/luma.oled #320: Migration from Python 3.7 to 3.9 gone bad

majuss, 2020-12-15:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.
Hi guys,
I'am in the process of upgrading my project from Python 3.7 to 3.9 but the Luma.oled library is spazzing out.. In the attached picture you see a scrolling text. With 3.7 everything works fine but with Python 3.9 the scrolling text is completely broken. Minimal static examples all work.
RPI 3B+

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.
5.4.79-v7+

Python3.9.1 completely fresh venv

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.
Scrolling text without artifacts.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).
Scrolling text with haevy glitches.

[image]

I will update this issue with my findings.

thijstriemstra, 2020-12-15:
can you doublecheck if the installed package versions for 3.7 vs 3.9 are the same?

majuss, 2020-12-15:
```
luma.core                        2.2.0
luma.oled                        3.8.1
Pillow                           8.0.1
```
Just tested it with the same versions in Pyhton3.7. No problems there.

rm-hull, 2020-12-15:
Hi @majuss , which oled driver is it?
I experienced some odd glitches recently with pillow > 7.1.2 & python 3.8 when using the ssd1309 driver.
Could you try python 3.9 with pillow 7.1.2 and report back please?

majuss, 2020-12-16:
I use a SSD1322 and no same glitches with 7.1.2.

thijstriemstra, 2020-12-16:
smbus2 recently had a big update as well, but probably unrelated to glitches..

majuss, 2021-01-23:
Tried again with most recent packages. Exactly the same :(

thijstriemstra, 2021-01-24:
@majuss so it doesn't work with Pillow 8.x on Python 3.9 but Pillow 7.1.2 on Python 3.9 is ok? And Python 3.7 is always ok?

> Tried again with most recent packages. Exactly the same :(

Please specify which ones and version nrs.

majuss, 2021-01-24:
Python3.9 never works pillow version doesn't matter. Python3.7 is always okay.

rm-hull, 2021-01-26:
Well the interesting thing here is that the tests target python 3.6, 3.7, 3.8 and 3.9, and while they generally substitute the interface (I2C / SPI) for a mock, they do check to make sure the correct sequence of bytes is being produced for all the different displays across all versions of Python (albeit on Ubuntu / intel rather than Raspian / arm)

So that leads me to the following questions:
* How have you installed Python 3.9 ? From source or a specific binary?
* Which OS are you using?
* Have you tried Python 3.8?

Everything you've said (and not said so far) points to it being a local issue, so we really need to know more about your setup

majuss, 2021-01-26:
I use the latest version of Raspbian on a RPI 3B+.  Didn't tried Python3.8 yet. I installed Python directly from source.

majuss, 2021-09-01:
I switched to a Pi 4 and now everything works with Python3.9

majuss, 2023-10-30:
@rm-hull The exact same behaviour is back again with Python 3.11 latest raspberry pi os bookworm.

thijstriemstra, 2023-10-31:
@majuss what version of Pillow are you using? Can you test with latest (10.1.0)?

majuss, 2023-12-04:
@thijstriemstra how can I debug this? Even the runner example etc. is showing these artefacts again :/

majuss, 2024-07-14:
The pillow version doesnt matter. The bug is there even with 9.4

rm-hull, 2024-07-14:
Can you reproduce it on other devices?