rm-hull/luma.oled #324: Improve text visibility

fabocode, 2021-03-04:
#### Issue 
I started to work with this module and wanted to test graphics in the `NHD-2.7-12864WDW3-M` it's a 128x64 display and I did the following code to setup custom fonts and I found it here #247. 
```
from PIL import ImageFont
FontTemp = ImageFont.truetype("./TestFont.otf",16)
draw.text((0, 0), "Example Text", font=FontTemp)
```
#### Type of Raspberry Pi
Raspberry Pi 4 

#### Linux Kernel version
`Linux raspberrypi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux`

#### Expected behaviour
A better visibility and smooth letter writing in the display 

#### Actual behaviour
The problem is that the expected font's visibility is not the best and I want to know how I could improve this without having to use the legacy text used in the module like `LCD_FONT, SINCLAIR_FONT`. In smaller displays like the adafruit SSD1306 works great but this one since use the same resolution I think is when the problem starts to grow.... Do you have any idea about how to improve the text visibility?

I tested all the fonts included in the `luma.examples` repository but none of them improves the results.

[image]

rm-hull, 2021-03-04:
You dont say which driver class you are using ... lets presume `ssd1325` as I suspect that your device is a greyscale display based on the image, and so Pillow is trying to anti-alias when drawing the font. You can force it to be monochrome (mode=1 rather than RGB) when you initialize the class, e.g. something like:

```python
device = ssd1325(serial_interface=spi, mode="1")
```

Issue #280 may also give you some background context.

fabocode, 2021-03-04:
Hi and sorry about that!, You're right. I'm actually using the `ssd1322_nhd` driver. 

Thanks to your suggestion, I did have an improvement. 

```p
self.device = ssd1322_nhd(self.serial, mode="1", width=128, height=64)
```

[image]

closed 2021-03-04 by fabocode

rm-hull, 2021-03-04:
Glad you got it working! 👍