rm-hull/luma.oled #325: Double cleanup when using with `gpiozero.Button`

matemaciek, 2021-03-07:
_First of all thanks for all the work put in `luma.oled` and the whole `luma`. I discovered it yesterday and it was a real pleasure to work with it._

When using both `gpiozero.Button` and `luma.oled` I noticed that exit of the program was not clear. I managed to narrow down the problem to code below:
```
from luma.core.interface.serial import spi
from luma.oled.device import sh1106
Button(20)
sh1106(spi())
```
This results in stacktrace:
```
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 595, in _shutdown
    _devices_shutdown()
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 589, in _devices_shutdown
    Device.pin_factory.close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 87, in close
    super(RPiGPIOFactory, self).close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 95, in close
    pin.close()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 137, in close
    GPIO.cleanup(self.number)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
```

#### Type of Raspberry Pi
`Pi Zero`

#### Linux Kernel version
`Linux raspberrypi 5.4.83+ #1379 Mon Dec 14 13:06:05 GMT 2020 armv6l GNU/Linux`

#### Expected behaviour
Clean exit

#### Actual behaviour
If I'm debugging it right, double cleanup of `GPIO`

thijstriemstra, 2021-03-07:
gpiozero is not officially supported, use rpi.gpio instead.

matemaciek, 2021-03-07:
I digged deeper and it turned out serial interface is to eager in cleaning up GPIO, please take a look at https://github.com/rm-hull/luma.core/pull/237

closed 2021-03-08 by rm-hull