rm-hull/luma.oled #328: Font Awesome icons not aligned

EDIflyer, 2021-06-26:
#### Type of Raspberry Pi
Pi Zero W

#### Linux Kernel version
5.10.17+

#### Expected behaviour
Font Awesome icons to be horizontally and vertically aligned with each other

#### Actual behaviour
I found the Font Awesome v4 supplied in the examples repo didn't include some icons I was wanting to use, I therefore downloaded the latest free Font Awesome v5.15.3 (from https://fontawesome.com/v5.15/how-to-use/on-the-web/setup/hosting-font-awesome-yourself).

However I'm finding that some of the icons are misaligned. This happened a little bit with the previous version but seems even more marked now: (please ignore the random text, was just writing some gobbledygook!)

_'i' icon all lined up:_
[image]

_'cloud upload icon' all lined up:_
[image]

_however when combined they appear horizontally offset:_
[image]

_Same issue with icons on bottom row when they change:_
[image]

_'cloud upload' and 'home' icons seem moved across and arrow is a bit high
[image]

From the appearance on their website they should line up OK...
[image]
[image]

I could look at manually trying to change the x/y co-ordinates for each icon but would make it much trickier when replacing them.
Current code is:
```
        draw.text((112, 3), icon1, font=icon_font, fill="white")
        draw.text((112, 18), icon2, font=icon_font, fill="white")
        draw.text((112, 33), icon3, font=icon_font, fill="white")
```

rm-hull, 2021-06-26:
I think the usual way to handle this in Pillow is to work out what the text size is using https://pillow.readthedocs.io/en/stable/reference/ImageDraw.html#PIL.ImageDraw.ImageDraw.textsize and then you should be able to center the icons programmatically.

For example, you could create a helper method like here: https://github.com/rm-hull/TR4-monitor/blob/master/tr4_monitor/common.py#L6-L8

EDIflyer, 2021-06-27:
Many thanks, @rm-hull - was totally unaware of that function! 😮 👍 

It definitely helped to flag up the difference - for example the sizes (x, y) for the icons in the 3rd image in my post above were as follows...
```
icon1 size:  (5, 13)
icon2 size:  (12, 12)
icon3 size:  (15, 12)
```
I took your advice to create a helper method and now call it for each icon. I have slightly bigger ones for the menu at the bottom so give them target_size of 14 vs 12 on the ones at the end of each line.
```
def centre_text(draw, text, font, x_offset, y_offset, target_size):
    icon_x=x_offset+(target_size-draw.textsize(text, font)[0]) / 2
    icon_y=y_offset+(target_size-draw.textsize(text, font)[1]) / 2
    draw.text((icon_x, icon_y), text, font=font, fill="white")
```
The result isn't perfect (when dealing with a 128x64 display I guess you get to the point where a single pixel makes a difference so there's a limit to how much granularity is possible!) but definitely much improved!

[image]

Thanks again for your help!

closed 2021-07-17 by EDIflyer

EDIflyer, 2024-02-17:
Warning - for anyone wondering why this doesn't work anymore! - it turns out Pillow have decided to remove this `textsize` attribute in v10!  (See [their docs](https://pillow.readthedocs.io/en/stable/releasenotes/10.0.0.html#font-size-and-offset-methods))

Have just spent 2h trying to work out why my code was failing on a new version of Pi OS and has taken a while to work out this was the problem! - was thinking when it was bombing out it was due to some GPIO issue (there was also an error "Falling back from lgpio: No module named 'lgpio'" but looks like it can just be ignored - sent me on a bit of a wild goose chase though!)

The closest replacement is `draw.textbbox` [[docs]](https://pillow.readthedocs.io/en/stable/reference/ImageDraw.html#PIL.ImageDraw.ImageDraw.textbbox) however you need to pass in extra co-ordinates and read different values from the array...

OLD
``` python
with canvas(device) as draw:
	draw.textsize("testing1234", text_font)
```
Which returns `(width, height)`

NEW
``` python
with canvas(device) as draw:
	draw.textbbox([0,0],"testing1234", text_font)
```
Which returns `(left, top, right, bottom)` - in this case `(0, 2, 56, 11)` - I therefore just used array items 2 & 3 rather as replacements for 0 & 1 previously.

If you only want the width then `draw.textlength` will do that (in this case it will return `56.0`).