rm-hull/luma.oled #330: I also need RST when using I2c on some of my displays

DJ-Dingo, 2021-07-22:
On some of my SPI / i2c Oled displays (ssd1306 / sh1106) i made a hardware shift on the back, by moving a resistor so it now run I2c instead of SPI. But on some of these cheap 6pins Oled displays, it still needs to use RST to work.

I can probably make it work with a 100nf capacitor but would now rather just be able to control it from my code, just like in SPI. Since it is for a plugin and not all users are capable of doing that kind of technical thing.

Would it be possible to add the gpio_RST and gpio_DC functions for use under I2c as well?

If I now add etc. gpio_RST to my I2c line, the display does not work, they are just black.

CRImier, 2021-07-30:
Hey! Why did you close this? Seems like an issue that'd affect other people, too.

DJ-Dingo, 2021-07-30:
Hey @CRImier, sorry for that. I reopen it.

It seems like the only solution right now is to do some tricks with the RES pin. Not shure if the gpio_RST and gpio_DC functions will help here alone. The SPI on theese 6pin OLEDs are not working on my Raspberry pi, even that it is a SPI device.

So what i did to make it work with I2c instead, was doing some hardware changes on the back. to make it a I2c OLED.
Then since the RES-pin still needs a low pulse at startup and high voltage during operation (as in a normal SPI mode), a 100nF capacitor connected to GND from the RES, and a 10Kohm resistor from RES to VCC is doing the job for now. 

[image]

closed 2023-06-11 by thijstriemstra