rm-hull/luma.oled #339: Trying to make a fake terminal

TeCoEd, 2022-03-24:
#### Expected behaviour
I am trying to build a code that displays letters on the screen as you type them.  I can do this with the code below.  However, when I get to the end of the first line, which is about letter 20 I need the next letters to start on the next line down, and then after this line is full, the new letters appear on the third line.

Any ideas / examples how to do this?

Thanks

#### Actual behaviour
When I enter letter 20, the whole line moves down each time I enter a new letter.

https://www.youtube.com/shorts/zKJp5TXxIGw

Code

    try:
        #print('alphanumeric key {0} pressed'.format(key.char))
        letter = (key.char) # single letter
        print (letter)
        terminal_list[list_num][pos] = letter

        with canvas(device) as draw:

            #program needs to remember previous letters
            total_letters = str(terminal_list) ### remove symbols
            
            #display the letter
            pos =pos + 1
            draw.text((list_num, pos), total_letters, fill="white")

            #testing
            print ("item number", pos)
            print ("list number", list_num)
    
            
            print (terminal_list)
            if list_num is not 3: #change for number of lines may need a fake line?
                if pos == 20: #change to match the width of the screen
                    list_num = list_num + 1
                    pos = 0
                elif pos == 20:
                    list_num = list_num + 1
                    pos = 0    
                else:
                    pass
            else: # reset and start again
                # or clear the screen?
                list_num = 0
                pos = 0

thijstriemstra, 2022-04-18:
@TeCoEd if you are using `luma.core.virtual.terminal`, set `word_wrap` to `True`

https://github.com/rm-hull/luma.core/blob/e4e3623d2e374bde152f4703ff535c90e35dc232/luma/core/virtual.py#L215

TeCoEd, 2022-05-15:
Thanks for all the support, code updated.

closed 2022-05-15 by TeCoEd