rm-hull/luma.oled #340: SSD1351 128x128 Color Issue

yuniq-neko, 2022-04-18:
#### Type of Raspberry Pi
Raspberry Pi 4B (4GB)

#### Linux Kernel version
Linux RPi-Dash-Cluster 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

#### Expected behaviour
RGB = RGB

#### Actual behaviour
RGB ≠ RGB

Red and Blue seem to have swapped, resulting in the palette becoming "BGR".

yuniq-neko, 2022-04-18:
Photo: [image]

rm-hull, 2022-04-18:
If you look here (https://luma-oled.readthedocs.io/en/latest/api-documentation.html#luma.oled.device.ssd1351) at the API docs, there is a `bgr=` flag you can set to true. This should resolve the issue.

yuniq-neko, 2022-06-19:
Apologies on the delay on responding to this, adding --bgr does indeed fix the issue. :)

closed 2022-06-19 by yuniq-neko