rm-hull/luma.oled #342: SSD1106 vs SH1106

felice64, 2022-04-25:
Thank you guys for the good work you do.
I am quite confused about SH1106 and SSD1106
I have few displays SSD1106 and I am not sure if the driver used for SH1106 is compatible.
I would appreciate some direction on what house to drive the SSD1106.

Thanks a million

felice64, 2022-04-28:
I did use the driver SH1106 on the LCD marked SSD1106 and looks like is working fine. I can display text and lines without problems. The only big issue I have is to display pictures. I am not sure if the workflow/code  I am using is wrong or if the issue is with the formatting of the picture.
I have not find in the examples a straight example on how to display an image on the display. All examples with pictures have complexities such as rotation, bouncing, etc.  I would appreciate a few lines of code  to test and some info about how to format and display a  picture for a 128x64 display.
Thanks for your support.

thijstriemstra, 2023-01-11:
> I did use the driver SH1106 on the LCD marked SSD1106 and looks like is working fine. 

Great, closing.

closed 2023-01-11 by thijstriemstra