rm-hull/luma.oled #361: White display SPI issue with 6.x kernels

jolla, 2023-01-23:
After updating kernel to any 6.x kernel intermittently the display does not work, it will illuminate but is solid white, if I downgrade back to 5.19 kernel then the display works fine.

Display: st7735
Interface: SPI

1.44inch LCD Display HAT For Raspberry Pi, 128x128 Pixels, SPI Interface
https://www.waveshare.com/1.44inch-lcd-hat.htm 

Example tested:
python3 3d_box.py --interface spi --height 128 --display st7735 

#### Type of Raspberry Pi
CM4

#### Linux Kernel version
Linux wlanpi-f06 6.1.7-v8-wlanpi+ #1 SMP PREEMPT Fri Jan 20 14:26:31 UTC 2023 aarch64 GNU/Linux

#### Expected behaviour
Image should be displayed

#### Actual behaviour
Display is solid white

thijstriemstra, 2023-01-23:
RaspberryPi OS doesn't support 6.x (yet) afaik: https://en.wikipedia.org/wiki/Raspberry_Pi_OS#Releases

Not sure what wlanpi is?

jolla, 2023-01-25:
I was able to resolve this issue by switching the interface type from `spi` to `gpio_cs_spi` and then defining the chip select pin using the `--gpio-chip-select` parameter.