rm-hull/luma.oled #367: Array is incorrectly shifted for 96x16 OLEDs.

gsmecher, 2023-03-15:
#### Type of Raspberry Pi

N/A - using an FTDI USB dongle.

The display is an AliExpress special like [this one](https://www.aliexpress.com/item/1005002862936939.html).

#### Linux Kernel version

N/A

#### Expected behaviour

With the following code:

```
serial = luma.core.interface.serial.ftdi_i2c(device='ftdi://ftdi:232h/1', address=0x3c)
device = luma.oled.device.ssd1306(serial, width=96, height=16)
with luma.core.render.canvas(device) as draw:
    draw.rectangle(device.bounding_box, outline="white", fill="black")
    draw.text((15, 4), "Hello World", fill="white")
```

...I expect a box that occupies the full canvas area, with text in the middle.

#### Actual behaviour

...the first 16 columns are black, and the last 16 columns are shifted off the end of the display:

[image]

I can correctly center the display with the following hack:

```
diff --git a/luma/oled/device/__init__.py b/luma/oled/device/__init__.py
index 012a80a..579b07b 100644
--- a/luma/oled/device/__init__.py
+++ b/luma/oled/device/__init__.py
@@ -255,7 +255,7 @@ class ssd1306(device):
         self._pages = height // 8
         self._mask = [1 << (i // width) % 8 for i in range(width * height)]
         self._offsets = [(width * (i // (width * 8))) + (i % width) for i in range(width * height)]
-        self._colstart = (0x80 - self._w) // 2
+        self._colstart = (0x60 - self._w) // 2
         self._colend = self._colstart + self._w
 
         self.command(
```

This results in the following:

[image]

I am guessing there are some hardware assumptions about how the OLEDs are attached to the SSD1306 that are violated by this module.

rm-hull, 2023-03-15:
So it will be already correct for the vast majority of devices .. except yours. I have found there to be many quirks with what are the "same" chipsets.. so the key here would be to retain existing behaviour but allow that parameter to be overridden for cases such as yours.

gsmecher, 2023-03-15:
Are there 64x16 devices in the wild that follow the existing _colstart math? If so, then definitely, those devices absolutely need to keep working, and I can draft a PR that allows overrides.

gsmecher, 2023-03-15:
Actually, I think this is a regression introduced [in 2017](https://github.com/gsmecher/luma.oled/commit/3ae59770ff2238c4fc9103bbeea17c089db71eae) and my PR should correct it.

The 2017 commit added the _colstart logic in question. Before it, 96x16 displays were supported but would have had an equivalent _colstart of 0. After it, 96x16 displays would have used a _colstart of 16.

thijstriemstra, 2023-03-15:
>  the key here would be to retain existing behaviour but allow that parameter to be overridden for cases such as yours.

Agreed. Ideally, this would be tested with and without ftdi_i2c.

gsmecher, 2023-03-15:
@thijstriemstra - I can put together something that exposes _colstart as a parameter if you like, but I think this issue (and the linked PR #368) are pure regression fixes that do not justify a configurable parameter or API changes on their own.

PR #368 (2023-03-16): Fix 96x16 OLED panel offset
This code has been tested on 128x32 and 96x16 panels. I am only just ducking into this repository - please review with a jaundiced eye to make sure it makes sense and is correct.

fixes #367
closed 2023-03-16 by rm-hull