rm-hull/luma.oled #380: from oled.device import ssd1306, sh1106 ModuleNotFoundError: No module named 'oled'

StefCoders, 2023-08-31:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.
 RPi 4 2gb

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

Linux piaware 5.15.76-v7l+ #1597 SMP Fri Nov 4 12:14:58 GMT 2022 armv7l GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

To just run.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

   from oled.device import ssd1306, sh1106 
ModuleNotFoundError: No module named 'oled'

thijstriemstra, 2023-08-31:
try: from luma.oled.device import ssd1306, sh1106

StefCoders, 2023-08-31:
Worked but.. Traceback (most recent call last):
  File "/home/pi/adsb.py", line 38, in <module>
    from luma.oled.device import ssd1306, sh1106
  File "/home/pi/.local/lib/python3.9/site-packages/luma/oled/device/__init__.py", line 38, in <module>
    from luma.core.device import device, parallel_device
  File "/home/pi/.local/lib/python3.9/site-packages/luma/core/device.py", line 9, in <module>
    from PIL import Image
  File "/home/pi/.local/lib/python3.9/site-packages/PIL/Image.py", line 82, in <module>
    from . import _imaging as core
ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory

StefCoders, 2023-08-31:
The python code is https://github.com/jprochazka/adsb-receiver/blob/master/build/display/current_total-aircraft_with-positions.py

StefCoders, 2023-08-31:
Traceback (most recent call last):
  File "/home/pi/adsb.py", line 39, in <module>
    from oled.render import canvas
ModuleNotFoundError: No module named 'oled'

thijstriemstra, 2023-08-31:
you need to install the Pillow system dependencies, try: sudo apt-get install python3-pillow

thijstriemstra, 2023-09-11:
did you manage to resolve it @StefCoders?

StefCoders, 2023-09-17:
Yep.

closed 2023-09-17 by StefCoders