rm-hull/luma.oled #387: Semi-random pixels on SSD1322 - how to fix?

TheoVDBerge, 2024-02-06:
Hello, 

First of all thank you for this great library! It is working great so far.

However, I am having once issue recently, which I will explain below.

#### Type of Raspberry Pi
Raspberry Pi Zero WH

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.
Linux raspberrypi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux

My screen is displaying semi-random pixels, akin to the troubleshooting reference on Read The Docs. However, the screen worked normally prior to this, so I know I am using the right driver. Is there a way to fit this? Reset the screen in a way? For example, when running the following code, the screen remains the way it was:

```from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.oled.device import ssd1322

serial = spi(device=0, port=0, bus_speed_hz=8000000)

device = ssd1322(serial, width=256, height=64)

device.clear()

try:
    while True:
        with canvas(device) as draw:
            draw.rectangle(device.bounding_box, outline="white", fill="black")
            draw.text((30, 20), "Updating...", fill="white")
except KeyboardInterrupt:
    pass

finally:
    device.cleanup()
```

Do you have any idea how to resolve this?

Thanks in advance!

[image]

thijstriemstra, 2024-02-06:
> However, the screen worked normally prior to this,

When? how? under what (other)circumstances?

TheoVDBerge, 2024-02-07:
Thanks for your reply and apologies for the lack of details!

So, I have a SSD1322 that I have connected using a 4SPI protocol to my Raspberry Pi Zero WH. I have wired it according to the documentation [here](https://luma-oled.readthedocs.io/en/latest/hardware.html#spi), for which many thanks again.

Using this set-up I have been able to print some text on the screen just fine, hence this is what I consider 'prior' to the above case. However, (I think) what caused this issue is that during prototyping I moved around the screen a bit, which caused one of the jumper cables (I believe it was the power-in jumper cable from the Raspberry Pi) to temporarily disconnect. Since then the screen has been stuck on semi-random pixels and a bit of what I previously had programmed to be displayed on the screen.

I also would like to highlight that I do not think there is anything wrong or missing with the library, but maybe due to the cable temporarily disconnecting something went wrong in the memory or internals? Given this GitHub page, I was hoping you (or someone else) might have run into a similar issue before and would know how to 'reset' the screen, as I did not find much information about it and I am relatively new to tinkering with such electronics.

Thank you again!

thijstriemstra, 2024-02-07:
> what caused this issue is that during prototyping I moved around the screen a bit, which caused one of the jumper cables (I believe it was the power-in jumper cable from the Raspberry Pi) to [...]

You tried taking off the power and restarting?

>  , but maybe due to the cable temporarily disconnecting something went wrong in the memory or internals?

Sounds likely.

TheoVDBerge, 2024-02-08:
As silly as it sounds, taking off the power and restarting it was one of the first things I tried to do. Maybe I plugged it back in too quickly, because I just tried to run nit just now (after letting it sit unplugged since I wrote this post), and it is back to normal again.

Sorry and thanks for you time!

closed 2024-02-08 by TheoVDBerge