rm-hull/luma.oled #391: luma.oled - device on SPI fails with BOOKWORM

FdeAlexa, 2024-06-29:
First of all: thanks for the LUMA solutions.
I am trying to run some python program with the new software platform RaspiOS 12 (bookworm) on the raspberry pi4.
All the programs runs correctly on 11.9 (bullseye) on several OLED on I2C and SPI, but with the new OS there is no way to run on SPI.
I have tryed some luma.examples too and the results are, unfortunatly, negative.
Would you please be so kind to help me on this issue?
Best regards and thanks,
Francesco

thijstriemstra, 2024-06-30:
> the results are, unfortunatly, negative.

What exactly? It should run on bookworm without any problems.

FdeAlexa, 2024-06-30:
I have tryed a python program that runs on bullseye but not on bookworm even 3d_box.py (for example) does not run on SSD1309 on SPI. Other programs on ili9341 (on SPI) run correctly.
Best regards,
Francesco

inclementweather, 2024-09-29:
I may be having this issue too, there's just no output on the screen.  I'm using a ssd1306 connected to a RPi Zero 2W with SPI on Pi OS bookworm.  Initially I followed the wiring [here](https://learn.adafruit.com/monochrome-oled-breakouts/python-wiring) and the display worked with a test script using adafruit_ssd1306 library.  Tried switching to luma and tested with the emulator and it worked, but on the actual hardware I get no output.  I rewired to match the pinout [here](https://luma-oled.readthedocs.io/en/latest/hardware.html#spi) but I still get no output.  The test script using adafruit's library still works after I change the pin settings to match the updated wiring.  There's no apparent errors.  I haven't tried bullseye.

kingjkj, 2024-10-09:
Hi, I have the same problem here.
I'm using Moode Audio which is based on Bookworm and I cannot make my OLED SSD1309 work in SPI with my Raspberry 5.
I'm using lgpio. I tried gpiozero and that's the same.
I always have these kind of errors.
I really don't understand what's wrong.

> Traceback (most recent call last):
  File "<string>", line 56, in <module>
  File "<string>", line 41, in main
  File "<string>", line 20, in init_oled
  File "/usr/local/lib/python3.11/dist-packages/luma/core/interface/serial.py", line 302, in __init__
    bitbang.__init__(self, gpio, transfer_size, reset_hold_time, reset_release_time, DC=gpio_DC, RST=gpio_RST)
  File "/usr/local/lib/python3.11/dist-packages/luma/core/interface/serial.py", line 190, in __init__
    self._DC = self._configure(kwargs.get("DC"))
  File "/usr/local/lib/python3.11/dist-packages/luma/core/interface/serial.py", line 203, in _configure
    self._gpio.setup(pin, self._gpio.OUT)
RuntimeError: Cannot determine SOC peripheral base address

kingjkj, 2024-10-09:
In file serial.py giving me errors, I cant read :

`:param gpio: GPIO interface (must be compatible with `RPi.GPIO <https://pypi.org/project/RPi.GPIO>`__).`

So I guess that luma cannot dirve spi oled on RPi 5 as RPI.GPIO is not compatible and I have to use LGPIO.
Am I right ?
Is there a workaround ?